eclipse-threadx/threadx #229: Cortex M4 SMP mode support

AJIOB, 2023-02-20:
**Is this request related to a particular hardware platform, SoC, board? Please describe.**

Some SoCs, such as [TI AM572x](https://www.ti.com/product/AM5728), has Dual-Cortex M4 (IPUx blocks on diagram).

**Describe why you are asking for this support?**

TI supports Cortex M4 SMP mode with their TI-RTOS, but it is not pre-certified.

**Additional context**

TI AM572x block diagram:
[image]

goldscott, 2023-02-21:
Hi @AJIOB - we do not yet have Cortex-M SMP. Is there a reason you need SMP on Cortex-M cores?

AJIOB, 2023-02-22:
Hello @goldscott. As I know, that is a single possibility to use both Cortex M4 for every IPU block.

fdesbiens, 2026-09-09:
Hi @AJIOB, apologies for the very long silence on this one.

Short answer: Eclipse ThreadX still has no Cortex-M SMP port, and there is none on the roadmap today. Every SMP port in the tree targets Cortex-A, Cortex-R8, MIPS interAptiv or ARC HS, plus the Linux and Win64 simulators.

The reason is structural rather than a matter of effort. The ThreadX SMP layer leans on a set of primitives that Armv7-M does not define, so each one would have to be reinvented per SoC: a core identity register (the Cortex-A ports read MPIDR with `MRC p15, 0, r0, c0, c0, 5` in `_tx_thread_smp_core_get`), a global exclusive monitor shared between the cores so that the `LDREX`/`STREX` spinlock in `_tx_thread_smp_protect` actually arbitrates rather than succeeding locally on both cores, barriers and a defined memory model for the shared kernel data structures, and an inter-core interrupt so `_tx_thread_smp_core_preempt` can force a reschedule on the other core. On a dual Cortex-M4 subsystem such as the AM572x IPU every one of those is SoC-specific glue rather than an architectural guarantee, which is also why TI's SMP support is tightly coupled to their own RTOS and their own devices. Nothing in the newer Cortex-M generations changes that either, since Armv8-M and Armv8.1-M still have no MPIDR and no architected inter-core interrupt.

That said, I do not want to leave you with a flat no, because the need behind your request is a real one. If what you are after is MPU-based, hard real-time, safety-oriented multicore, Arm's own answer to that is Armv8-R rather than Cortex-M: Cortex-R52 and Cortex-R82 do provide MPIDR, GIC software-generated interrupts and a cluster-level global exclusive monitor, so they satisfy the SMP layer's requirements architecturally instead of per SoC. We already ship a uniprocessor Cortex-R52 port in `ports/cortex_r52`, and we already maintain a 32-bit, MPU-based SMP port in `ports_smp/cortex_r8_smp`, which operates without hardware cache coherency between cores by keeping the shared kernel structures in shared, non-cacheable memory. An R52 SMP port is therefore a realistic thing to build in a way that a Cortex-M SMP port is not, so I have opened #710 to track it and credited you there for raising the underlying use case.

For the AM572x specifically, though, the approach we would recommend today is AMP rather than SMP: run an independent ThreadX instance on each Cortex-M4, partition the workload statically between them, and use shared memory plus the device's hardware mailboxes for IPC. If pre-certification is what is really driving the request, that is also the pragmatic path, since each instance stays a plain single-core ThreadX, whereas a new Cortex-M SMP port would be freshly written code with no certification history behind it for a long time.

I am closing this one as not planned, since a Cortex-M SMP port is not something we intend to take on. Please follow #710 if Armv8-R SMP would work for you.

closed 2026-09-09 by fdesbiens