eclipse-threadx/threadx #231: Need to understand about ports and ports_module folders difference

kaushaleinfo, 2023-03-02:
Hi Team,
I need to understand ports and ports_module folder difference in latest version of ThreadX.
Basically I am using arm cortex-m3 and IAR IDE and I need to pick some port related file but I confused Which one should I take ?
Could you guys tell what are difference in both folders
[image]

Thanks 
Kaushal Panchal

goldscott, 2023-03-02:
'ports' contains all the port-specific code for 'normal' ThreadX. 'ports_module' is only for [ThreadX Modules](https://learn.microsoft.com/en-us/azure/rtos/threadx-modules/chapter1).

kaushaleinfo, 2023-03-03:
I am upgrading ThreadX_5.5 version to ThreadX_6.2.0 version and controller is stm32f3xxx which is ARM_cortex_m3 and I use IAR IDE for project.
Could you guys help us which folder should I use?

Thanks 
Kaushal Panchal

goldscott, 2023-03-03:
Use ports.

closed 2023-03-03 by goldscott