eclipse-threadx/threadx #232: Could not find __tx_SVCallHandler definition

kaushaleinfo, 2023-03-04:
Hi Team,

I am using ThreadX 6.2.0 but could not find definition of **__tx_SVCallHandler**  however we are using it in application layer, So definition of this handler is must in ThreadX library files.
Could you guys please help us where is __tx_SVCallHandler definition.

Thanks 
Kaushal Panchal

goldscott, 2023-03-04:
What processor and compiler are you using?

kaushaleinfo, 2023-03-04:
I am using Stm32f2xx controller which based on ARMcortex_m3, I use IAR IDE for this project.

wangwen-4220, 2023-07-05:
@kaushaleinfo , there is no need to define __tx_SVCallHandler  in  6.2.0 ThreadX, we have sample projects to demonstrate how to use 6.2.0 threadx : https://github.com/azure-rtos/samples

fdesbiens, 2026-09-11:
Hi @kaushaleinfo, and apologies for the very long delay in getting back to you. I am the ThreadX project lead at the Eclipse Foundation, and I am going through the backlog of open issues.

`__tx_SVCallHandler` is not part of the ThreadX kernel. It only exists in the ThreadX **module manager** ports, where the `SVC` instruction is the mechanism a user-mode module thread uses to transition back into the kernel. For the Cortex-M3 IAR port you will find its definition in `ports_module/cortex_m3/iar/module_manager/src/tx_thread_schedule.s`, where it is published alongside `SVC_Handler` and decodes the SVC parameter to dispatch the kernel entry, the user-mode return, and the module dispatch calls.

The plain Cortex-M3 kernel does not use `SVC` at all. Context switching is done entirely through PendSV and SysTick, so the only handlers `ports/cortex_m3/iar/src/tx_thread_schedule.s` publishes are `PendSV_Handler` and `__tx_PendSVHandler`. That is why there is no `__tx_SVCallHandler` anywhere in the 6.2.0 kernel sources for your configuration, and why @wangwen-4220 was right that you do not need one.

In the IAR example build the SVCall vector slot points at `SVC_Handler`, which `ports/cortex_m3/iar/example_build/cstartup_M.s` declares `PUBWEAK` and resolves to the default infinite-loop handler. The example builds for the other toolchains (AC5, AC6, GNU, GHS) do define a symbol called `__tx_SVCallHandler` in their `tx_initialize_low_level` file, but it is only a placeholder to fill the vector table slot: its entire body is `B __tx_SVCallHandler`. So even there it is a stub with no functionality.

If your application layer references `__tx_SVCallHandler`, it was almost certainly copied from a module manager example or from a vendor SDK that ships its own startup file. For a non-module application you have two options: either remove the reference and let the weak `SVC_Handler` from `cstartup_M.s` cover the vector slot, or provide your own `SVC` handler if your application genuinely uses the `SVC` instruction for something of its own.

I am relabelling this as a question rather than a bug and closing it, since there is nothing to fix in the kernel. Please feel free to reopen it or to open a new issue if you are actually trying to use ThreadX modules and are hitting a problem with the module manager port.

closed 2026-09-11 by fdesbiens