eclipse-threadx/threadx #258: Getting 'empty' message from tx_queue_receive

SchutzSeb, 2023-05-03:
Hi,

We are currently using ThreadX (6.2.0) along with Tracealyzer (4.6.6) in order to collect and visualize traces.
The software is running on an ATSAMS70 (GCC arm toolchain) and we have many threads running with different priority levels. Each thread has a dedicated queue in order to receive messages from another threads or ISRs.

We have observed that when the device is heavily loaded while running Tracealyzer tool we can get sometimes 'empty' message out from tx_queue_receive function. The investigations we have made up to now show that this happens because the thread that was suspended while waiting for a new message is being resumed while no message has been sent to its queue.

Here is the screenshot from Tracealyzer with some annotations:

[image]

We can see on one hand that /nonsil/ProcInterControllerComm is resumed while no new message has been sent to its queue, and on the other hand that /system/ProcHalControl is not resumed while a message has been sent to its queue (it was also suspended waiting for a new message). Moreover /system/ProcHalControl is more prioritary than /nonsil/ProcInterControllerComm.

We hope that you might have some suggestions to understand what is happening and how we could fix it properly.

Best regards.

xiuwencai, 2023-06-01:
Hi @SchutzSeb, could you please check the tx_thread_state value of the mistakenly resumed thread?

xiuwencai, 2023-06-07:
Hi @SchutzSeb, could you please share more information about your configuration: Have you defined anything in tx_user.h or defined any ThreadX configuration when compiling? Are you using the data cache? If so, is it configured to write-through or write-back mode? What's the version of the GCC?

SchutzSeb, 2023-06-07:
Hi @xiuwencai,

Yes we have some definitions into our tx_user.h:

```
#define TX_TIMER_TICKS_PER_SECOND 1000
#define TX_MAX_PRIORITIES 32
#define TX_ENABLE_STACK_CHECKING
#define TX_THREAD_USER_EXTENSION VOID* tx_thread_local_storage;
#define TX_ENABLE_EVENT_TRACE
```

The data cache is enabled on the microcontroller (`__DCACHE_PRESENT` set to 1) but I have found no additional configuration regarding this feature, so I'm not sure for the write mode (microcontroller is ATSAMS70).

Currently we use `gcc-arm-12.2.MPACBTI-Rel1-mingw-w64-i686-arm-none-eabi` for compiling.

williamelamie, 2024-02-05:
Was this issue ever resolved?

SchutzSeb, 2024-02-06:
We have found a way to avoid the issue to occur by disabling interrupts when calling tx_queue_send.
Although, as of today, we still have not found the origin of the problem.

williamelamie, 2024-02-06:
In theory, ThreadX will disable interrupts briefly inside the
tx_queue_receive/tx_queue_send.  When you get a chance, it would be
interesting to turn off the interrupt lockout you added and set a BP (or
step into) the tx_queue_send/receive code to ensure the interrupts are
disabled where they should be. It has the feel that interrupts are getting
enabled somewhere (maybe the tracing code) when they shouldn't be.

SchutzSeb, 2024-02-14:
Hello,
Thanks for your reply.
I have put assertion in tx_queue_send everywhere before TX_RESTORE is called to check if interrupts were enabled back while they should not. But no assertion occured.
Nevertheless I think you pointed in the good direction. I have tried to disabled any manipulation of the interrupts by Tracealyzer and it seems that the issue was not happening anymore (of course I had removed my fix before). The traces recorded were chaotic but the device could run fine.
Now at least I have an idea where the issue is possibly coming from.
Thanks for your help !
I'll give update in case I get new information.

fdesbiens, 2026-09-11:
Hi @SchutzSeb, and my apologies for the very long delay in coming back to you. I am Frédéric Desbiens, the Eclipse Foundation project lead for Eclipse ThreadX. We are working through the older issues, and this one deserves a real answer rather than a close. I think we can now name the root cause precisely, and I believe your instinct in February 2024 was exactly right.

Start with the data point you gave @xiuwencai, because it is the whole story in one number. You observed `_tx_thread_current_ptr -> tx_thread_state == 5`. That is `TX_QUEUE_SUSP` (`common/inc/tx_api.h:131`). A thread that is actually executing must be `TX_READY`, which is 0. A running thread whose control block still says `TX_QUEUE_SUSP` cannot be produced by any ordinary path through the queue services, so this was never a logic error in `tx_queue_send`/`tx_queue_receive`. It is the signature of a suspend or resume sequence that was interrupted part-way through.

Here is the window that matters. `_tx_thread_system_resume()` sets `tx_thread_state = TX_READY` at `common/src/tx_thread_system_resume.c:167`, inside a critical section that runs from the `TX_DISABLE` at line 104 to the `TX_RESTORE` at line 501. `_tx_thread_system_suspend()` has the same shape, with `TX_DISABLE` at `common/src/tx_thread_system_suspend.c:113` and the first `TX_RESTORE` at line 357. Removing the thread from the ready list, updating the priority bit maps and updating `_tx_thread_execute_ptr` must all be atomic with respect to interrupts.

Now notice what ThreadX calls from inside those windows. `TX_TRACE_IN_LINE_INSERT()` is invoked at `tx_thread_system_suspend.c:172` and at `tx_thread_system_resume.c:130`, that is, with interrupts already masked. You have `TX_ENABLE_EVENT_TRACE` in your `tx_user.h`, so those hooks resolve into the Percepio recorder. Any code reached from there must treat the interrupt state as strictly save-and-restore.

The recorder shipped with Tracealyzer 4.6.6 does not. In `kernelports/ThreadX/trcCTI.c` at line 1967 the PRIMASK save slot is declared `static TRACE_ALLOC_CRITICAL_SECTION()`, a single file-scope variable rather than an automatic local. `xTraceEventBeginRawOffline()` performs `TRACE_ENTER_CRITICAL_SECTION()` and then returns with interrupts still masked, and the matching `TRACE_EXIT_CRITICAL_SECTION()` lives in a completely different function, `xTraceEventEndOffline()`. On Cortex-M those macros expand to `xTraceCriticalSectionStatus = __get_PRIMASK(); __set_PRIMASK(1);` and `__set_PRIMASK(xTraceCriticalSectionStatus);`. Because the saved value is shared mutable state held across a two-function span rather than a per-invocation local, any other enter/exit pair that runs in between overwrites it, and the outer exit then writes back a PRIMASK captured at an unrelated call site. At thread level that value is normally 0, which means interrupts get re-enabled in the middle of ThreadX's suspend or resume. An ISR calling `tx_queue_send()` in that window lands on a half-completed sequence, and you get precisely the two symptoms in your Tracealyzer screenshot: a lower-priority thread resumed with nothing in its queue, and a higher-priority thread left suspended although a message was sent to it.

Percepio appears to have fixed this. The `static` slot is present in 4.6.6, 4.7.0 and 4.8.2, and is gone from 4.9.0 onwards. In 4.9 and later the same file uses a function-local `TRACE_ALLOC_CRITICAL_SECTION();` with the enter and exit confined to a single function, wrapping only the `xTraceCheckThreadSwitch()` call. So the concrete recommendation is to move to Tracealyzer 4.9 or newer, retest without your `tx_queue_send` interrupt workaround, and report back if anything remains.

This also explains every observation you made along the way. Your assertions before each `TX_RESTORE` in `tx_queue_send` found nothing because the damage happens in the suspending thread's own path, not in the sender. Masking interrupts around `tx_queue_send` closed the window from the other side. And most tellingly, disabling Tracealyzer's manipulation of the interrupts made the problem stop.

Two loose ends for completeness. The data cache on the ATSAMS70 is not a factor: the ThreadX control structures are touched only by the single Cortex-M7 core, so there is no coherency exposure unless DMA writes into that memory. And `TX_ENABLE_STACK_CHECKING`, `TX_MAX_PRIORITIES` and `TX_THREAD_USER_EXTENSION` in your `tx_user.h` are all benign here.

Since the defect is in the trace recorder rather than in ThreadX, I am closing this one. Please do reopen it if you can reproduce the corruption on a current recorder, or with ThreadX's own stock trace hooks and no third-party recorder in the picture, and we will take another look.

closed 2026-09-11 by fdesbiens

SchutzSeb, 2026-09-15:
Hi,

Thank you for this detailed answer, it has been really helpful.

Best regards.