eclipse-threadx/threadx #265: Get time in milliseconds

MaJerle, 2023-05-21:
**Is your feature request related to a problem? Please describe.**

I'm always frustrated when I need to implement (for every project) custom time to ms conversion with 
```c
#define tx_time_get_ms() (tx_time_get() * (1000 / TX_TIMER_TICKS_PER_SECOND))
```

**Describe the solution you'd like**
I propose we add this macro as part of standard API of ThreadX, next to declaration of `tx_time_get()` function.

**Describe alternatives you've considered**
None - Application simply requires `ms` time for further calculation, even if its resolution is not at `1ms` but depends on the `TX_TIMER_TICKS_PER_SECOND` info.

**Additional context**
Add any other context or screenshots about the feature request here.

MaJerle, 2023-05-21:
I can create a pull-request, should you agree to add this feature.

TiejunMS, 2023-05-22:
Thanks @MaJerle! We will discuss internally and get back to you.

TiejunMS, 2023-06-02:
@MaJerle , we discussed internally and decided not to add this macro to ThreadX API. The main reason is that due to the resolution of ThreadX timer, it may confuse users require 1ms precision. Instead, some of user applications will add additional timer in the system to get 1ms precision.

In your case, I recommend adding this macro to tx_user.h which can be shared by multiple projects as a workaround.

closed 2023-06-08 by TiejunMS