eclipse-threadx/threadx #266: RH850 series porting

kvvipa0028, 2023-05-21:
Do we currently have product migration for RH850 series? I'm trying to use THREADX on a RH850 device.
My E-mail is [kvvipa002@163.com]

TiejunMS, 2023-05-23:
Shared V850 port through OneDrive. Please check.

closed 2023-05-23 by TiejunMS

drdaliang, 2023-08-18:
Hi @TiejunMS , I am interested in RH850 ports too. Would you mind provide me too? My email is drdaliang@gmail.com

TiejunMS, 2023-08-18:
> Hi @TiejunMS , I am interested in RH850 ports too. Would you mind provide me too? My email is [drdaliang@gmail.com](mailto:drdaliang@gmail.com)

Shared V850 port through OneDrive. Please check.

trinhdoanh, 2024-08-19:
Hi @TiejunMS,

Would you kindly please share me RH850/V850 porting?
I have a project which needs ThreadX working on Renesas RH850 MCU.
My email [trongtribk06@gmail.com](mailto:trongtribk06@gmail.com)

Thank you