eclipse-threadx/threadx #283: Why are you using void * for module instance and such in modules thread extention

amgross, 2023-07-12:
For example here void * is used here:
https://github.com/azure-rtos/threadx/blob/8ff9910ddc089e7af91e4089b26f225aba5d7eb7/ports_module/cortex_a35/gnu/inc/tx_port.h#L214
I know that the module struct wasn't declared yet, but it can be solved by struct forward declaration.

I think that using the void* instead of the struct pointer is misleading.

williamelamie, 2024-02-05:
That is worth looking into... since it isn't a published API, it might be something that could be changed.

fdesbiens, 2026-09-09:
Thanks for raising this, and apologies for the very long delay.

You are right that `struct` forward declaration would make this compile, so the ordering problem is not the whole story. The reason these members are `VOID *` is layering rather than declaration order: `TX_THREAD` is defined by the kernel in `tx_api.h`, while `TXM_MODULE_INSTANCE` is owned by the module manager in `common_modules/inc/txm_module.h`, which includes `tx_api.h` rather than the other way round. Making the kernel control block name a module-manager type would mean every one of the 32 module port headers has to forward declare a type it otherwise knows nothing about, and the kernel would carry a reference to a layer it does not depend on.

There is also a compatibility cost. `TX_THREAD` is part of the published API surface, so changing a member from `VOID *` to `struct TXM_MODULE_INSTANCE_STRUCT *` is a source-breaking change for any application, debugger awareness plugin or middleware that reads or assigns those members today. The same argument applies to `tx_thread_module_entry_info_ptr` and to the `tx_queue_module_instance`, `tx_semaphore_module_instance` and `tx_event_flags_group_module_instance` members, so it would not be a one-line change.

I agree the current form is less self-documenting than it could be, and the module manager sources do cast to `TXM_MODULE_INSTANCE *` at every use, so the intent is at least consistent. Given the breadth of the change and the compatibility risk against a fairly modest readability gain, I am going to leave the types as they are and close this. If you feel strongly about it, a PR against `dev` would be considered on its merits — but it would need to cover all the module port headers and all the affected object control blocks at once.

closed 2026-09-09 by fdesbiens