eclipse-threadx/threadx #288: Unable to create static variable in module

stdiehl, 2023-07-17:
Hey,

I have tried to create a static variable in a module content but the variable gets initialized with a random number from anywhere. 

I have tried it with a STM32F407-DISCO and with a MIMXRT1062-EVKB with Release 6.1.1, 6.1.10 and 6.2.1 of ThreadX.

For the STM controller I build it with the CubeIDE, for MIMXRT1062 with MCUXpressoIDE and with cmake in VSCode with gcc-arm-none-eabi-10.3-2021.10 - same result.

My example for the CubeIDE is attached.

The Memory config is:
- Module Manager:
MEMORY
{
  RAM               (xrw)  : ORIGIN = 0x20000000,  LENGTH = 0x0000F000
  FLASH             (rx)   : ORIGIN = 0x08000000,   LENGTH = 0x00020000
}
- Module:
MEMORY
{
  FLASH (rx)  : ORIGIN = 0x08020000,  LENGTH = 0x00020000  
  RAM   (rwx) : ORIGIN = 0x20010000, LENGTH = 0x00010000
}

Module Flags:
- GCC Assembler: -fpie -fno-plt -mpic-data-is-text-relative -msingle-pic-base
- GCC Compiler: -mthumb -fpic -fno-plt -mpic-data-is-text-relative -msingle-pic-base
- Linker: -nostartfiles -nodefaultlibs -nostdlib -e _txm_module_thread_shell_entry

What I did not find is, where the register r9 is set before the gcc_setup.S of the module gets executed? In my case, r9 is set to 0x20000c00 but for my understanding, r9 must be set to 0x20010000. Where comes 0x20000c00 from?
In flash memory, the values of the static variables are set correctly but the copy of the flash data to RAM is not correct.

What am I doing wrong? Is there a flag missing or is there a problem with the load of the module from the module manager?

Many thanks for any help and BR

[cubeide.zip](https://github.com/azure-rtos/threadx/files/12069501/cubeide.zip)

williamelamie, 2024-02-05:
How is the module being loaded via the module manager?  Generally, ThreadX modules should be built in a PIC and PID fashion so the R9 data area address should be determined when the module is loaded.

hefanbo, 2024-09-03:
> What I did not find is, where the register r9 is set before the gcc_setup.S of the module gets executed?

`_tx_thread_stack_build`

> for my understanding, r9 must be set to 0x20010000. Where comes 0x20000c00 from?

No, r9 is not related to 0x20010000 at runtime. When loading the module, the system allocates a range of RAM for the module to use. r9 is the PIC base address related the the allocated RAM at runtime.

> In flash memory, the values of the static variables are set correctly but the copy of the flash data to RAM is not correct.

If you were checking the values with some debugging tool, then go on reading. The addresses of static/global variables in a module are relocated. If your debugging tool does not claim to support ThreadX, it is likely that you are looking into wrong addresses.

fdesbiens, 2026-09-09:
Hello @stdiehl, and my apologies for the very long delay in responding. This is not a ThreadX defect: the problem is in your module compiler flags.

Your module is compiled with `-mpic-data-is-text-relative`. The ThreadX sample module build scripts deliberately use the opposite, `-mno-pic-data-is-text-relative` — see `ports_module/cortex_m4/gnu/example_build/build_threadx_module_sample.bat` and `build_threadx_module_library.bat`, where every single compilation unit carries `-fpic -fno-plt -mno-pic-data-is-text-relative -msingle-pic-base`. Note that `-mpic-data-is-text-relative` is the *default* for bare-metal `arm-none-eabi`, so it has to be turned off explicitly.

Here is what the flag does to a static variable, with `arm-none-eabi-gcc -O1 -mcpu=cortex-m4 -mthumb -fpie -fno-plt -msingle-pic-base` on `static volatile int v; int get(void) { return v; }`:

```
-mpic-data-is-text-relative       -mno-pic-data-is-text-relative
  ldr  r3, [pc, #4]                 ldr   r3, [pc, #8]
  add  r3, pc                       ldr.w r3, [r9, r3]
  ldr  r0, [r3, #0]                 ldr   r0, [r3, #0]
```

In the first case the variable is addressed at a fixed displacement from the PC, so the module reads whatever happens to sit at that offset from its code in flash. It never consults `r9` and never consults the GOT, so the relocation work done by `gcc_setup.S` at module start-up is simply bypassed. That is exactly the "random number from anywhere" you observed. In the second case the address is fetched from the GOT indexed by `r9`, which is the only model the ThreadX module manager supports.

On your question about `r9`: it is loaded from the initial thread context that `_txm_module_manager_thread_stack_build` builds (`ports_module/cortex_m4/gnu/module_manager/src/txm_module_manager_thread_stack_build.s:115`), and its value is `txm_module_instance_data_start`, which `_txm_module_manager_internal_load` allocates from `_txm_module_manager_byte_pool` (`common_modules/module_manager/src/txm_module_manager_internal_load.c:254,293`). That byte pool lives in the *module manager's* RAM, which in your configuration starts at `0x20000000` — hence `0x20000c00`. The `RAM ORIGIN = 0x20010000` in your module linker script is never a runtime address; the `__RAM_segment_start__` symbol is used only to compute section *offsets*, which `gcc_setup.S` then rebases onto `r9`. So `r9` is correct and is not supposed to equal `0x20010000`.

One further remark: unless you are using `txm_module_manager_absolute_load`, the module's data is always placed wherever the manager's byte pool has room, so the module must be fully position independent for both code and data. Also compare your linker script against `ports_module/cortex_m4/gnu/example_build/sample_threadx_module.ld` — the ordering of `.got`, `.data` and `__new_got_start__` there is what `gcc_setup.S` expects.

Please switch to `-mno-pic-data-is-text-relative` and let us know if the problem persists. I am closing this as answered, but do not hesitate to reopen if it does.

closed 2026-09-09 by fdesbiens