eclipse-threadx/threadx #290: Missing SYSTEM flag for PUBLIC include directories in CMakeLists.txt

MarkAtBosch, 2023-07-31:
Unfortunately, ThreadX 6.1.12 fixed issue #138 **CMake project include directories should be "system" ones** only partially. The problem was fixed in `common/CMakeLists.txt` but not in all other CMakeLists.txt files which publish parts of the ThreadX API.

Changes in [common/CMakeLists.txt](https://github.com/azure-rtos/threadx/commit/8c3c08f10826c4d5332a6d8e2fdf014b793a456a#diff-a77d393383a84e0dd42ebd261b80443390335377813ca160e32abf8c52effac2) at commit 8c3c08f10826c4d5332a6d8e2fdf014b793a456a:
```
@@ -202,6 +202,7 @@ target_sources(${PROJECT_NAME}

   # Add the Common/inc directory to the project include list
   target_include_directories(${PROJECT_NAME} 
+      SYSTEM
       PUBLIC 
       ${CMAKE_CURRENT_LIST_DIR}/inc
   )
```

At least for all CMakeLists.txt files in ports/*THREADX_ARCH*/*THREADX_TOOLCHAIN* the same fix is required to handle all ThreadX include directories as system includes and thus compiler warnings will be skipped.

MarkAtBosch, 2023-12-06:
Gentle ping ... @goldscott  @yuxin-azrtos 

FYI @ryanwinter @gdelazzari

williamelamie, 2024-02-05:
Hi Mark, Yuxin is with me over at rtosx.com (px5rtos.com)....  I'll ask him tomorrow about this.

williamelamie, 2024-02-23:
> Hi @williamelamie <https://github.com/williamelamie> and @yuxin-azrtos
> <https://github.com/yuxin-azrtos>, any news on this?

Hi Mark,
I'm sorry about that... Hope to have some feedback early next week.
Best regards,
Bill

williamelamie, 2024-02-27:
Hi Mark,

I've discussed this with Yuxin, and we believe what you proposed earlier in
the thread is correct. Is your request simply that it needs be updated in
the repository?

Best regards,

Bill

Please also feel free to reach out to me at ***@***.***

MarkAtBosch, 2024-02-27:
Hi @williamelamie,

Yes, it is still my recommendation to apply the fix from [common/CMakeLists.txt](https://github.com/azure-rtos/threadx/commit/8c3c08f10826c4d5332a6d8e2fdf014b793a456a#diff-a77d393383a84e0dd42ebd261b80443390335377813ca160e32abf8c52effac2) at commit 8c3c08f10826c4d5332a6d8e2fdf014b793a456a:
```
@@ -202,6 +202,7 @@ target_sources(${PROJECT_NAME}

   # Add the Common/inc directory to the project include list
   target_include_directories(${PROJECT_NAME} 
+      SYSTEM
       PUBLIC 
       ${CMAKE_CURRENT_LIST_DIR}/inc
   )
```

at least into the CMakeLists.txt files in ports/*THREADX_ARCH*/*THREADX_TOOLCHAIN*.

Essentially, the fix should be applied to all CMakeLists.txt files which publish parts of the ThreadX API.

Best regards,
Mark

MarkAtBosch, 2024-05-17:
Hi @williamelamie and @yuxin-azrtos, any news on this?

fdesbiens, 2026-09-09:
Hi @MarkAtBosch, and thank you for your patience on this one.

You were right, and the fix is now on its way in #713. I applied the `SYSTEM` keyword everywhere a ThreadX header is published, not only under `ports/<arch>/<toolchain>`: all 60 port and SMP port directories, the four architecture sources under `ports_arch` so the generated ports stay in step, the POSIX and FreeRTOS compatibility layers, the generated `tx_user.h` directory in the top-level `CMakeLists.txt`, and the parallel library build used by the SMP regression suite. Include directories that are private to an example or test build were left alone, since nothing is published from them.

I verified in the generated `build.ninja` that the port include directory is now passed as `-isystem` rather than `-I`, and the ThreadX, SMP and POSIX suites all still build and pass. The POSIX layer was worth a second look because it deliberately publishes replacements for `errno.h`, `pthread.h` and friends, but `-isystem` directories are still searched ahead of the toolchain's own, so that override keeps working.

I will close this issue once the pull request is merged.

fdesbiens, 2026-09-09:
Fixed in #713, which is now merged into `dev`. Thanks again for reporting this and for following up on it, @MarkAtBosch.

closed 2026-09-09 by fdesbiens