eclipse-threadx/threadx #292: 'cortex-m0' building fail

PING020903, 2023-08-10:
[image]

I try building 'cortex-m3' of files, but no error. I do not know how to read there '.s' files, because i do not understand. Where 'undefined reference' should be define in the project? Can you help me, please? My files download form this repository, also unmodified。

williamelamie, 2024-02-05:
I duplicated your problem. It seems that the startup code and predefined symbols for this version of ThreadX do not match the current GCC compiler generated symbols, specifically in the section of  cortexm0_crt0.S having to do with copying initialized sections into RAM.  For a quick proof of concept, I commented-out the following section in cortexm0_crt0.S and got a clean build:

/*  ldr r0, =__data_load_start__
  ldr r1, =__data_start__
  ldr r2, =__data_end__
  bl crt0_memory_copy
  ldr r0, =__text_load_start__
  ldr r1, =__text_start__
  ldr r2, =__text_end__
  bl crt0_memory_copy
  ldr r0, =__fast_load_start__
  ldr r1, =__fast_start__
  ldr r2, =__fast_end__
  bl crt0_memory_copy
  ldr r0, =__ctors_load_start__
  ldr r1, =__ctors_start__
  ldr r2, =__ctors_end__
  bl crt0_memory_copy
  ldr r0, =__dtors_load_start__
  ldr r1, =__dtors_start__
  ldr r2, =__dtors_end__
  bl crt0_memory_copy
  ldr r0, =__rodata_load_start__
  ldr r1, =__rodata_start__
  ldr r2, =__rodata_end__
  bl crt0_memory_copy
*/

This may not be the real solution, but at least it identifies the problem!

fdesbiens, 2026-09-09:
Hi @PING020903, and apologies for the very long delay in getting back to you.

Your report was correct, and @williamelamie's diagnosis was on the right track: the problem was not in `cortexm0_crt0.S` but in the linker script beside it. `cortexm0_crt0.S` references 23 linker script symbols and `sample_threadx.ld` defined only 12 of them, so `__text_start__`, `__text_end__`, `__text_load_start__`, the `.rodata` and `.fast` section symbols, and the `.ctors` and `.dtors` load addresses were all unresolved. Commenting out the copy loops removed the references, which is why that worked as a proof of concept, but it also removed the startup copy of initialised sections.

This is now fixed on `dev` by #595. The Cortex-M0 linker script has been brought to the same shape as the Cortex-M4 one, which defines all 23 symbols, including a `.fast` section with no content whose symbols exist so that the startup copy is a no-op. The same change also added `.syntax unified` to `cortexm0_crt0.S` so the file assembles with LLVM as well as GNU.

I have just rebuilt `ports/cortex_m0/gnu/example_build` from a clean `dev` checkout with `build_threadx.sh` followed by `build_threadx_sample.sh`, and both complete without errors. The Cortex-M0 example is also now covered by our CI, so this will not regress silently again.

Closing this as fixed. Thank you for the report, and thank you for your patience.

closed 2026-09-09 by fdesbiens