eclipse-threadx/threadx #293: Question: tx_thread_suspend() behavior on a suspended thread

jamesEmerson112, 2023-08-12:
Hi. I am seeking to confirm a specific behavior of a thread within ThreadX when it is suspended, particularly how it interacts with new suspension requests.

The scenario in question involves a thread that is already suspended to wait for a message from the queue, followed by a new request to suspend the thread (tx_thread_suspend()), and then the arrival of a new message. I am trying to understand whether the thread would be awakened to process the new message and then get suspended by the new request, or if the new message would be queued until the original suspension is lifted. 

Additionally, I am interested in how this behavior interacts with different synchronization mechanisms such as event flags, mutexes, and semaphores, and how these mechanisms might influence the thread's ability to process new messages and respond to additional suspension requests.

My assumption is that if a thread that is suspended to wait for a new message from a queue gets tx_thread_suspend(), the tx_thread_suspend() will be held internally until the original thread is resumed due to a new message; however, once it is awaken to be resumed, the held request tx_thread_suspend() may immediately take affect, causing it unable to process the new message until tx_thread_resume() is called. Is this behavior expected?

amgross, 2023-08-13:
Hi @jamesEmerson112 ,
see https://github.com/azure-rtos/threadx/blob/bc8bed494d3a37eea022c1385008bdab991f3370/common/src/tx_thread_suspend.c#L188-L193
It seems like calling tx_thread_suspend on already suspended thread has no effect.

(I am not from threadx team, so you may want to wait for their response to ensure)

jamesEmerson112, 2023-08-13:
> It seems like calling tx_thread_suspend on already suspended thread has no effect.

Thanks, @amgross . I suppose my follow up question is: "how do we suspend a thread, which may be in a different state of suspension, so that the thread can and only can be resumed by tx_thread_resume(thread) or a method that invokes it?"

jamesEmerson112, 2023-08-13:
I spent some more time to read the documentation regarding my question at
[    {

        /* Just set the delayed suspension flag.  */
        thread_ptr -> tx_thread_delayed_suspend =  TX_TRUE;

        /* Set status to success.  */
        status =  TX_SUCCESS;
    }](https://github.com/azure-rtos/threadx/blob/bc8bed494d3a37eea022c1385008bdab991f3370/common/src/tx_thread_suspend.c#L195C1-L203C1)
    
Is it safe to say that if a thread is suspended on a queue of messages, when it receives a new message, as long as tx_thread_delayed_suspend is set TRUE, the thread will be awaken, but get suspended immediately again due to that flag before processing the new message, am I correct?

amgross, 2023-08-14:
Now that you are raising it. I think my previous answer was wrong.

In case for example the system suspended on queue, the state won't be TX_SUSPENDED as I claimed but:
https://github.com/azure-rtos/threadx/blob/bc8bed494d3a37eea022c1385008bdab991f3370/common/src/tx_queue_receive.c#L444
Hence when you will call tx_thread_suspend you will probably exactly will get to:
https://github.com/azure-rtos/threadx/blob/bc8bed494d3a37eea022c1385008bdab991f3370/common/src/tx_thread_suspend.c#L194-L202
And will be suspended till resume will called (I don't sure which conditions needed, I think will need both to call resume and to send message on the queue)

TiejunMS, 2023-09-07:
Here is the note from the API description for [tx_thread_suspend](https://learn.microsoft.com/en-us/azure/rtos/threadx/chapter4#tx_thread_suspend)().

> If the specified thread is already suspended for another reason, this suspension is held internally until the prior suspension is lifted. When that happens, this unconditional suspension of the specified thread is performed. Further unconditional suspension requests have no effect.

closed 2023-10-15 by jamesEmerson112