eclipse-threadx/threadx #294: Different processor variants in CM7 build config for IAR

Corderbollie, 2023-08-15:
**Describe the bug**
ARM7TDMI is set in the 'release' build configuration of the example build for IAR CM7 tx project instead of Cortex M-7.
This applies to the actual master and at least version 6.2.1.

**To Reproduce**
Steps to reproduce the behavior:
1. Check 'General Options > Processor variant > Core' in the options of the tx release build configuration.
1. Trying to compile the release target will cause license errors in IAR Embedded Workbench with Cortex-M License installed.

**Expected behavior**
Release target of tx project should compile with IAR Cortex-M License installed.

**Impact**
Annoyance

Corderbollie, 2023-08-15:
Now I noticed that none of the release build targets in the workspace is functional.
Why is that?

TiejunMS, 2023-09-07:
You are right. The *release* configurations are not set for those example projects. Please switch to *debug* configuration for demo purpose.

closed 2023-09-07 by TiejunMS