eclipse-threadx/threadx #298: Assessment of the difficulty in porting CPU architecture for the threadx

wangyuliu, 2023-09-11:
Hello everyone! I am working on implementing a tool to assess the complexity of CPU architecture porting. It primarily focuses on RISC-V architecture porting. In fact, it can be estimated on average across various architectures.My focus is on the overall workload and difficulty of transplantation in the past and future.As part of my dataset, I have collected the **threadX** project. **I would like to gather community opinions to support my assessment. I appreciate your help and response!** Based on scanning tools, the porting complexity is determined to be moderate, with a moderate amount of code related to the CPU architecture in the project. (referring to the overall workload from adapting the project to a specific architecture to achieving full functionality on that architecture). Is this assessment accurate? I look forward to your help and response.

williamelamie, 2024-02-05:
Not sure if this is what you are looking for, but ThreadX support for most processor architectures is composed of 8 assembly language functions in separate files (typically tx_*.s) and one C include file tx_port.h.  The design is for minimal hardware dependency and ease of porting.  You can find examples in the "ports" folder of the repo.

fdesbiens, 2026-09-22:
Your “moderate” assessment is reasonable, with one important distinction.

For a basic single-core CPU and compiler port, ThreadX deliberately keeps the architecture-specific boundary small. A typical port provides tx_port.h and a small set of assembly routines for context switching, interrupt handling, scheduling, stack construction, and the timer tick. The code volume is modest, although implementing it correctly requires detailed knowledge of the ABI, privilege and interrupt model, stack frames, and optional floating-point state.

“Full functionality” is broader than the kernel port, however. It also requires startup and linker support, exception vectors, timer and interrupt-controller integration, board drivers, runtime-library support, build tooling, and hardware or emulator validation. Much of that work now belongs in [SampleX](https://github.com/eclipse-threadx/samplex), where the [CORE-V MCU](https://github.com/eclipse-threadx/samplex/tree/dev/OpenHW/CORE-V-MCU) and [Microchip PolarFire SoC](https://github.com/eclipse-threadx/samplex/tree/dev/targets/Microchip/POLARFIRE_ICICLE_RENODE) examples illustrate substantially more ambitious target integrations.

I would therefore classify the core ThreadX CPU port as moderate difficulty with a relatively small amount of architecture-specific code. Complete platform enablement can range from moderate to high difficulty depending on the SoC, board, toolchain, validation environment, and whether features such as SMP or ThreadX Modules are required.

closed 2026-09-22 by fdesbiens