eclipse-threadx/threadx #309: Hard fault on _tx_byte_pool_search function

carlochuawee, 2023-10-25:
Hi,

I am having a hard fault exception when I call tx_byte_allocate. 

This only happens after some time of byte allocate and release. There are still plenty of available bytes according to the pool but somehow current_pointer points to an address outside the scope of the pool and memory that is not available for my chip. After further investigation, I found the issue to be in _tx_byte_pool_search.

The current pointer variable (current_ptr) tries to access memory 0x2492bc7e. This is my data at address 0 when dereference. That memory is not available for H753 and obviously will throw a hard fault error trying to access this memory.

Specifically on line below inside the do while(examine_blocks != ((UINT) 0)) loop

work_ptr = TX_UCHAR_POINTER_ADD(current_ptr, (sizeof(UCHAR *)));
free_ptr = TX_UCHAR_TO_ALIGN_TYPE_POINTER_CONVERT(work_ptr);
if ((*free_ptr) == TX_BYTE_BLOCK_FREE) //<-----------------------------------------HARD FAULT

I can't upload images here with debug screenshots of variables. But see the detailed post below.

https://learn.microsoft.com/en-us/answers/questions/1403352/azure-rtos-byte-pool-problem

For reference only, I am using STM32H753 and Azure RTOS version 6.2.0.

Thanks

TiejunMS, 2023-10-26:
It sounds like a buffer overflow issue. Could you check the usage of memory allocated from byte pool? If the issue can be reproduced, set a data breakpoint may catch the root cause.

carlochuawee, 2023-10-27:
Hi Tiejun,
I am swamped with another task today. Anyway, I will investigate this further and hopefully will get back to you early next week when I reproduce it.
Thanks

carlochuawee, 2023-11-07:
Hi Tiejun,

It seems to be as you stated - a buffer overflow in the application due to porting with another File system. I will be closing this issue as this is not an Azure RTOS problem.
Thanks.

closed 2023-11-07 by carlochuawee