eclipse-threadx/threadx #313: Add WFE in tx_thread_schedule in SMP system

carll00226, 2023-11-10:
HI, support team:
        I want to add WFE when running in the loop of _tx_thread_schedule in SMP threadx system.  and  release SEV when task ready, would you please tell me where to add the WFE and SEV instruction?

williamelamie, 2024-02-05:
The first place to look is the TX_ENABLE_WFI logic. An example can be found in the ThreadX Cortex-A5x SMP ports. The wait here is released by _tx_thread_smp_core_preempt via the macro TX_THREAD_SMP_WAKEUP_LOGIC.  Of course, this logic could be modified to meet your needs.

fdesbiens, 2026-09-22:
The idle wait belongs in the port-specific `_tx_thread_schedule`, after the core’s `_tx_thread_execute_ptr` entry has been checked for `TX_NULL` and interrupts have been re-enabled. The Armv8-A SMP scheduler sources show this loop under `TX_ENABLE_WFI`; a `WFE` implementation should follow the same placement.

The wake-up should use the existing `TX_THREAD_SMP_WAKEUP_LOGIC` / `TX_THREAD_SMP_WAKEUP(i)` hook. ThreadX calls this after assigning a runnable thread to another core, so `SEV` does not need to be added to every thread-ready path. The wake-up routine should include an appropriate barrier, such as `DSB ISHST`, before `SEV` so the execute-pointer update is visible before the event is sent.

Please retain the port’s inter-processor interrupt path as well. `SEV` wakes cores blocked in `WFE`, but it does not preempt a core that is already executing a lower-priority thread; that still requires `_tx_thread_smp_core_preempt()`.

Since this is an old issue, I am closing it for now. If you need further help, please reopen it and specify which SMP processor and compiler port you are using.

closed 2026-09-22 by fdesbiens