eclipse-threadx/threadx #315: RISCV fpu context save and restore according to the MSTATUS.FS

xuzihan351, 2023-11-15:
RISCV has a FS filed in MSTATUS, it indicates the status of the fpu. It can be initial, clean, dirty of off. When dealing with the fpu context save and restore, the FS filed may help to increase the preformence of the context switching. When the MSTATUS.FS is clean or off, the context saving and restoring can be skip.
Is it possible to implement this feature in the future?

williamelamie, 2024-02-05:
This is definitely possible...  tx_thread_schedule.s, tx_thread_system_return.s, tx_thread_context_save.s, and tx_thread_context_restore.s are the likely places for the enhanced logic.

Alexbruliau, 2024-05-29:
Did you find out any issues related to FPU when porting to RISC-V ?

xuzihan351, 2024-05-30:
No issues found at the moment

Alexbruliau, 2024-05-30:
Not event this kind of https://github.com/eclipse-threadx/threadx/issues/389 ?

Alexbruliau, 2024-05-30:
thanks you targeted a mpfs250 board?

xuzihan351, 2024-05-31:
Not that. I am targeting HPMicro series boards