eclipse-threadx/threadx #317: debugging demo_threadx on WSL2

mendez23, 2023-11-17:
Hello!

On WSL2, threadx sample program demo_threadx compiles and runs ok. However I'm unable to debug the program -- after setting breakpoint and running the program, the IDE notifies that "Exception has occurred. User defined signal 1" and stops at
line 403 sigsuspend(&_tx_linux_thread_wait_mask); in void  _tx_linux_thread_suspend_handler(int sig) tx_initialize_low_level.c

If continued, the program runs few steps and then stops again..

1. Is there an explanation to this behavior?
2. Is there a recommended workaround so debug session can continue?

Thanks in advance!

TiejunMS, 2023-11-20:
You can use following gdb command to bypass the signals used by ThreadX for Linux port.
```
handle SIGUSR1 noprint
handle SIGUSR2 noprint
handle SIG32 noprint
```
Those signals are used to suspend and resume native pthread.

mendez23, 2023-11-20:
@TiejunMS it works; thank you!

closed 2023-11-20 by mendez23