eclipse-threadx/threadx #319: Does threadx have some tools, just like ASAN(address sanitizer) for tracking memory issues

frank-amy, 2023-11-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Does threadx have some tools, just like ASAN(address sanitizer) for tracking memory issues
https://source.android.com/docs/security/test/asan?hl=zh-cn

williamelamie, 2024-02-05:
I'm not aware of any such tool for ThreadX, but it is possible to write a function to interrogate the integrity of ThreadX objects, e.g., threads, queues, memory pools, etc.  Each control block has an ID field that can be examined for corruption and there is metadata between each allocated block in block and byte memory pools.

frank-amy, 2024-02-29:
thanks for your reply
Yes, maybe those functions can be public for every user, to do dianostic or tracking

fdesbiens, 2026-09-22:
I confirm that ThreadX does not currently provide a general AddressSanitizer-like facility. I opened #764 to define the investigation needed before this can become a scoped implementation. It covers the fault classes and environments to support, possible compiler- and ThreadX-based approaches, real-time and resource costs, diagnostics, testing, and documentation.

I labeled the investigation `help wanted` because contributors with ThreadX internals, embedded toolchain, or sanitizer-runtime experience can help establish a practical direction. No project resources are currently committed to either the investigation or an implementation. I suggest continuing any technical scoping discussion in #764.

closed 2026-09-22 by fdesbiens