eclipse-threadx/threadx #320: Ask for help

wgz-code, 2023-11-21:
I recently learned through studying the ThreadX system that the ThreadX system supports the Coterx-A9 architecture, because I saw Coterx-A9 in the Ports directory. I now plan to transplant the ThreadX system to the development of NXP imx6q This board is an ARM Cortex-A9  board. I would like to ask if you have relevant instruction manuals to support me in this work?

TiejunMS, 2023-11-22:
Could you please reach out to NXP to see if there is existing port? https://community.nxp.com/

williamelamie, 2024-02-05:
Taking the generic Cortex-A9 support to any specific processor of the same architecture is fairly straightforward. The biggest effort is to integrate the IRQ (and FIQ) interrupt handlers to match the ThreadX requirement. You will also need to setup a periodic timer interrupt (typical is 10ms) and modify the linker control file to make sure the proper address is being supplied to tx_application_define.

fdesbiens, 2026-09-22:
There is no i.MX6Q-specific board support package in this repository. I recommend starting with the generic [Cortex-A9 port](https://github.com/eclipse-threadx/threadx/tree/master/ports/cortex_a9) and its [port guide](https://github.com/eclipse-threadx/threadx/blob/master/ports/cortex_a9/gnu/readme_threadx.txt).

I expect the i.MX6Q integration to require board-specific startup and clock initialization, a suitable linker memory map, exception-vector placement, GIC interrupt dispatch, and a periodic timer interrupt. I would start with the uniprocessor port on one core; if the application needs multiple cores, I would use the corresponding Cortex-A9 SMP port as the basis instead. I also recommend consulting NXP’s i.MX6Q documentation and support community for the SoC-specific initialization.

I believe the earlier replies answer the original question, so I am closing this issue. I would be happy to examine a focused follow-up if a specific integration problem remains.

closed 2026-09-22 by fdesbiens