eclipse-threadx/threadx #337: Threadx on Microblaze soft processor

cncherisher, 2023-12-24:
**Is this request related to a particular hardware platform, SoC, board? Please describe.**
Hello

Is threadX supported on Xilinx Microblaze soft processor which use RISC-V 32 bit? I see there is support for RISC-V 32bit for IAR toolchain and RISC-V 64bit for GCC toolchain, but Vitis IDE use GCC toolchain, so is there any listed steps to port risc-v32 for GCC toolchain?

williamelamie, 2024-02-05:
The best approach is to convert the ThreadX RISC-V 32-bit support with IAR to GCC, which amounts to changing the assembly code (tx*.s), tx_port.h, and linker control file (first available memory address). This is easier than creating a new port from scratch, but still a good amount of work.

fdesbiens, 2026-09-22:
I can confirm that ThreadX now includes a RISC-V32 GNU port under [`ports/risc-v32/gnu`](https://github.com/eclipse-threadx/threadx/tree/master/ports/risc-v32/gnu), with build instructions and several example platforms. It was added in #490, so converting the IAR port is no longer necessary.

For MicroBlaze V, I would use that port as the starting point and adapt the platform-specific startup, linker layout, timer, interrupt-controller, and trap handling. I have not found a MicroBlaze V BSP or example in this repository, so I cannot claim that the port has been validated on that processor. Vitis also uses a `riscv64-unknown-elf-gcc` multilib toolchain, while the supplied ThreadX configuration expects `riscv32-unknown-elf-*`; that build configuration would need to be adjusted.

I would expect a complete MicroBlaze V BSP and sample project to be contributed to [SampleX](https://github.com/eclipse-threadx/samplex), while this repository retains the reusable architecture-port code.

If you meant the original MicroBlaze rather than MicroBlaze V, it does not implement RISC-V and this port does not apply. I am closing this question because the requested RV32/GNU port is now available; a separate feature request would be appropriate for a maintained MicroBlaze V BSP and sample.

closed 2026-09-22 by fdesbiens