eclipse-threadx/threadx #341: Why EPK kit is not supported in ThreadX release 6.3.0?

leonid-galperin, 2023-12-28:
**Describe the bug**
I'm using ThreadX port for Cortex-R5. I have several tasks in my application and  want to measure overall CPU load by means of kindly provided Execution Profile Kit 

According to _tx_api.h_ comments (line 618):
"For Azure RTOS 6, user shall use TX_EXECUTION_PROFILE_ENABLE instead of TX_ENABLE_EXECUTION_CHANGE_NOTIFY"
In fact there is no any use of TX_EXECUTION_PROFILE_ENABLE macros.

On the other hand _tx_thread_schedule.s_ contains lines:
```
#ifdef **TX_ENABLE_EXECUTION_CHANGE_NOTIFY**
@
@    /* Call the thread entry function to indicate the thread is executing.  */
@
    BL      _tx_execution_thread_enter      @ Call the thread execution enter function
#endif
```

Also there is not clear description how to use EPK to measure CPU load

williamelamie, 2024-02-01:
I see what you are seeing. It appears that the information in tx_api.h applies to building the C code. It appears that TX_ENABLE_EXECUTION_CHANGE_NOTIFY should be used when building the assembly code.  Have you tried that combination yet?

fdesbiens, 2026-09-22:
I reproduced this on `dev`: with `TX_EXECUTION_PROFILE_ENABLE`, the Cortex-R5 scheduler and interrupt paths emitted none of the EPK callbacks. I opened #766 to make every AC5, AC6, GNU, Green Hills and IAR hook accept the documented symbol while retaining compatibility with `TX_ENABLE_EXECUTION_CHANGE_NOTIFY`. I also added checks covering both configurations.

For Cortex-R5, I recommend defining `TX_EXECUTION_PROFILE_ENABLE` for the ThreadX and application builds, adding `utility/execution_profile_kit/tx_execution_profile.c` to the application, and adding that directory to the include path. I also recommend overriding `TX_EXECUTION_TIME_SOURCE` and `TX_EXECUTION_MAX_TIME_SOURCE` with a suitable free-running Cortex-R5 timer because the bundled default reads the Cortex-M DWT counter. I would calculate interval CPU load from the thread, ISR and idle time deltas as `100 * (thread + ISR) / (thread + ISR + idle)`.

Since this issue has been inactive for some time, I will close it for now. If the fix does not resolve the problem, please feel free to open a new issue with the details.

closed 2026-09-22 by fdesbiens