eclipse-threadx/threadx #346: Is it possible to register a notification callback when using tx_trace_user_event_insert?

rocketion, 2024-01-17:
Is it possible to register a notification callback when using tx_trace_user_event_insert?

williamelamie, 2024-02-01:
I don't see an easy way to do this without modifying the code in tx_trace_user_event_insert.c or the macro TX_TRACE_IN_LINE_INSERT.  Could you share a bit more on what you are trying to accomplish in the callback?

fdesbiens, 2026-09-22:
I confirmed that ThreadX does not provide a callback for each call to `tx_trace_user_event_insert()`. I would use `tx_trace_buffer_full_notify()` if notification is needed when the trace buffer wraps, including when a user event causes the wrap. If notification is needed after every user-event request, I would wrap `tx_trace_user_event_insert()` in application code and invoke the application callback there.

I am closing this as answered because the requested callback semantics were not clarified. I would be happy to reconsider a feature request that describes when the callback must run and what information it must receive.

closed 2026-09-22 by fdesbiens