eclipse-threadx/threadx #356: Potentially missing null check for thread pointer in _tx_mutex_get()

szsam, 2024-02-07:
The pointer `thread_ptr` is checked for null here:
https://github.com/eclipse-threadx/threadx/blob/d64ef2ab064047ab5aed15c7861e5a44ab8022e3/common/src/tx_mutex_get.c#L122-L168

But it is not checked in the following block:
https://github.com/eclipse-threadx/threadx/blob/d64ef2ab064047ab5aed15c7861e5a44ab8022e3/common/src/tx_mutex_get.c#L208-L395

billlamiework, 2024-02-19:
The rule is that tx_mutex_get cannot be called from a non-thread context with a suspension option. This is checked for in the error checking for tx_mutex_get, namely txe_mutex_get.  Error checking is enabled by default. If the application bypasses error checking it must be calling all APIs from the correct context.

As for the earlier check for thread_ptr non-NULL is to handle the call of tx_mutex_get from initialization (where thread_ptr is NULL). 

I hope that helps explain what is going on!

fdesbiens, 2026-09-03:
Thanks for the careful reading, @szsam, and my apologies for the very long delay in closing the loop here.

I went through the code again to confirm @billlamiework's explanation, and it holds. In the block you flagged, every dereference of `thread_ptr` sits on the `wait_option != TX_NO_WAIT` path; the `TX_NO_WAIT` path restores interrupts and returns `TX_NOT_AVAILABLE` without touching `thread_ptr` at all.

`thread_ptr` can only be `TX_NULL` when `_tx_mutex_get()` is reached from initialization or an ISR, and `_txe_mutex_get()` rejects any suspending request from those contexts with `TX_WAIT_ERROR` (it also rejects the timer thread when `TX_TIMER_PROCESS_IN_ISR` is not defined). Error checking is enabled by default, and an application that disables it takes on the responsibility of calling the API from a valid context — this is the documented contract for the whole `_tx_*` layer, not something specific to mutexes.

The earlier `if (thread_ptr != TX_NULL)` check exists because acquiring an uncontested mutex *is* legal during initialization, so that path must tolerate a null thread pointer.

I am therefore closing this as not a bug. If you have a reproducer that reaches a null dereference with error checking enabled, please reopen and I will take another look.

closed 2026-09-03 by fdesbiens