eclipse-threadx/threadx #358: Repeating error codes in common/inc/tx_api.h

niklass-l, 2024-02-13:
On lines 224 and 225, the error code 0x10 repeats:

#define TX_NO_MEMORY                    ((UINT) 0x10)
#define TX_START_ERROR                  ((UINT) 0x10)

Making it impossible to distinguish between these two error codes

amgross, 2024-02-14:
Hi @niklass-l 
Can you please add some more information?
Is this compile time error or runtime?
What is the exact error message/API etc.

niklass-l, 2024-02-14:
Hi @amgross, I'll elaborate.

It's a runtime error, it compiles just fine.

It's just that on this line: https://github.com/eclipse-threadx/threadx/blob/master/common/inc/tx_api.h#L224
The error code for TX_NO_MEMORY is 0x10.

But on the next line: https://github.com/eclipse-threadx/threadx/blob/master/common/inc/tx_api.h#L225
The error code for TX_START_ERROR is also 0x10.

That means the user of the library can't distinguish between these two error codes if they are ever returned from the same function.

Might not be a significant issue, just wanted to let you know.

hwmaier, 2024-02-14:
The only function which uses both error codes seems to be `txm_module_manager_external_memory_enable()` and indeed its impossible to distinguish which of the two errors caused this function to fail.

amgross, 2024-02-14:
Hi @niklass-l 
It was asked already at #176 and at #41 
Basically the answer from what I understand is we need to keep it as is for backwards compatibility.
It can be re-decided of course.

niklass-l, 2024-02-14:
Oh okay, apologies I didn't know that. Thanks!

closed 2024-02-14 by niklass-l

hwmaier, 2024-02-14:
I suggest to add a comment to the code which explains why they are numerically identical and that it stays for backward compatibility. Prevents from getting asked again...