eclipse-threadx/threadx #360: tx_thread_smp_protect

junomoto, 2024-02-16:
threadx/ports_smp/cortex_a7_smp/gnu/src/
tx_thread_smp_protect.S

you need using LDREX and STREX

@/**************************************************************************/
@/*                                                                        */
@/*       Copyright (c) Microsoft Corporation. All rights reserved.        */
@/*                                                                        */
@/*       This software is licensed under the Microsoft Software License   */
@/*       Terms for Microsoft Azure RTOS. Full text of the license can be  */
@/*       found in the LICENSE file at https://aka.ms/AzureRTOS_EULA       */
@/*       and in the root directory of this software.                      */
@/*                                                                        */
@/**************************************************************************/
@
@
@/**************************************************************************/
@/**************************************************************************/
@/**                                                                       */ 
@/** ThreadX Component                                                     */ 
@/**                                                                       */
@/**   Thread - Low Level SMP Support                                      */
@/**                                                                       */
@/**************************************************************************/
@/**************************************************************************/
@
@
@#define TX_SOURCE_CODE
@#define TX_THREAD_SMP_SOURCE_CODE
@
@/* Include necessary system files.  */
@
@#include "tx_api.h"
@#include "tx_thread.h"
@#include "tx_timer.h"  */

@/* Include macros for modifying the wait list.  */
#include "tx_thread_smp_protection_wait_list_macros.h"

    .global    _tx_thread_current_ptr
    .global    _tx_thread_smp_protection
    .global    _tx_thread_smp_protect_wait_counts
    .global    _tx_thread_smp_protect_wait_list
    .global    _tx_thread_smp_protect_wait_list_lock_protect_in_force
    .global    _tx_thread_smp_protect_wait_list_head
    .global    _tx_thread_smp_protect_wait_list_tail
    .global    _tx_thread_smp_protect_wait_list_size

    .arm
    .text
    .align 2
@/**************************************************************************/ 
@/*                                                                        */ 
@/*  FUNCTION                                               RELEASE        */ 
@/*                                                                        */ 
@/*    _tx_thread_smp_protect                          SMP/Cortex-A7/GNU   */ 
@/*                                                            6.1.12      */
@/*  AUTHOR                                                                */
@/*                                                                        */
@/*    William E. Lamie, Microsoft Corporation                             */
@/*                                                                        */
@/*  DESCRIPTION                                                           */
@/*                                                                        */ 
@/*    This function gets protection for running inside the ThreadX        */ 
@/*    source. This is ccomplished by a combination of a test-and-set      */ 
@/*    flag and periodically disabling interrupts.                         */ 
@/*                                                                        */ 
@/*  INPUT                                                                 */ 
@/*                                                                        */ 
[... 285 lines cut ...]
#endif
    B       _try_to_get_lock                    @ On waking, restart the protection attempt

_got_lock_after_waiting:
@
@    /* We're no longer waiting.  */
@    _tx_thread_smp_protect_wait_counts[this_core]--;
@
    LDR     r3, =_tx_thread_smp_protect_wait_counts @ Load waiting list
    LDR     r4, [r3, r1, LSL #2]                @ Load current wait value
    SUB     r4, r4, #1                          @ Decrement wait value
    STR     r4, [r3, r1, LSL #2]                @ Store new wait value value

@
@    /* Restore link register and return.  */
@
_return:

    POP     {r4-r6}                             @ Restore registers

#ifdef __THUMB_INTERWORK
    BX      lr                                  @ Return to caller
#else
    MOV     pc, lr                              @ Return to caller
#endif

billlamiework, 2024-02-19:
LDREX and STREX are being used for the locking mechansim. Are you suggesting additional locations?

junomoto, 2024-02-20:
Yes.
Three locations accesses "_tx_thread_smp_protection.tx_thread_smp_protect_in_force".
One lcation is using LDREX and STREX.
But, other two location aren't using.

billlamiework, 2024-02-20:
Since I last looked at this code some queuing logic was also added. In some of the other SMP port-specific code, the queuing logic was then removed. More investigation is needed to find the correct fix (which may include the removal of the queuing logic) as well as make sure _tx_thread_smp_protection.tx_thread_smp_protect_in_force is accessed properly. Are you actually experiencing a problem with the code, and do you currently have a fix?

junomoto, 2024-02-20:
I'm not good at English.
Sorry if I've made a mistake.

I found a problem.
TX_MPCORE_DEBUG_ENABLE is enabled.
Thread0 is fixed only core0.
Thread0 by core0 can't get _tx_thread_smp_protection.tx_thread_smp_protect_in_force, and enter a loop.
_tx_thread_smp_protection.tx_thread_smp_protect_thread is thread0.
But, _tx_thread_smp_protection.tx_thread_smp_protect_core is core1.

I predicted.
core0 got _tx_thread_smp_protection.tx_thread_smp_protect_in_force.
core1 waited getting it.
If core0 release it and try to get again at soon.
core0 and core1 gets it at same time.
So, they need using LDREX and STREX.
That code I pasted is already fixed.

billlamiework, 2024-02-21:
Your English is good, and I'm happy to hear everything is working. I recommend issuing a pull request with your fix so it gets properly reviewed.

yf13, 2024-09-26:
@junomoto, thanks for sharing this story about the SMP cortex-a7/gnu port.

Would you mind edit the first comment by enclosing the source code like below:

```
Contents of the source file

```

This way, it will be easier to view your modifications. I would have done this for you but unfortunately I don't have permissions for that.

fdesbiens, 2025-02-27:
@junomoto Did you end up submitting a PR or are you still interested in doing it? If so, I will keep this issue open. If not, I will close it in the next few weeks due to inactivity. Please let me know.

fdesbiens, 2026-09-22:
Thank you for reporting this, and I'm sorry it remained open for so long. I confirmed that the old Cortex-A7 SMP implementation did not use exclusive accesses in both queue-based acquisition paths, as you described.

I have since merged PR #707 into `dev`. In that change, I removed FIFO queueing from the Cortex-A5, Cortex-A7, and Cortex-A9 SMP ports and made every acquisition attempt use `LDREX`/`STREX`. This supersedes the proposed patch and removes the two non-exclusive paths you identified. I verified the GNU variants by assembly and disassembly, and CI passed, but I did not test the change on ARMv7-A SMP hardware.

I consider this fixed on `dev` and will close the issue. Please reopen it or file a new issue if you can still reproduce the problem with the current `dev` branch.

closed 2026-09-22 by fdesbiens