eclipse-threadx/threadx #369: _tx_semaphore_created_ptr is always NULL

xray-bit, 2024-03-08:
**Describe the bug**
I want to print the semaphore information, but the value of the pointer variable _tx_semaphore_created_ptr is always NULL.

Device: Cortext-M4
ThreadX Version: 6.4.1
Toolchain: arm-gnu-toolchain-13.2.Rel1-mingw-w64-i686-arm-none-eabi

**To Reproduce**
```c
#include <tx_api.h>
#include <tx_semaphore.h>

// ...

static void show_sem(void)
{
    uart_printf("sem_ptr = %p\n", _tx_semaphore_created_ptr);
    uart_printf("sem_cnt = %d\n", _tx_semaphore_created_count);
}

// ...
```
I have created several semaphores in other files and they are running normally.

**Expected behavior**
expected:
```
sem_ptr = 0x2000****
sem_cnt = [greater than 0]
```

but the actual output
```
sem_ptr = 0
sem_cnt = 0
```

amgross, 2024-03-14:
Hi @xray-bit , 
can you please give us 'minimal reproducible example'?
From what you wrote I would guess you have some bug and show_sem is running before you created the semaphores or after you deleted them.

xray-bit, 2024-03-21:
Sorry, I was busy earlier. Thanks for the reply.  
When I created a new project to reproduce the problem, it became normal again and I will check why the previous one was faulty.

closed 2024-03-21 by xray-bit