eclipse-threadx/threadx #370: The interrupt vector table offset register should not be set.

SeeDeer, 2024-03-08:
/* Setup Vector Table Offset Register.  */
  MOV     r0, #0xE000E000                         // Build address of NVIC registers
  LDR     r1, =_vectors                           // Pickup address of vector table
  STR     r1, [r0, #0xD08]                        // Set vector table address

  /* Set system stack pointer from vector value.  */
  LDR     r0, =_tx_thread_system_stack_ptr        // Build address of system stack pointer
  LDR     r1, =_vectors                           // Pickup address of vector table
  LDR     r1, [r1]                                // Pickup reset stack pointer
  STR     r1, [r0]                                // Save system stack pointer

in ports/cortex_m7/gnu/example_build/tx_initialize_low_level.S file, the interrupt vector table offset register is set, which will cause trouble for users. The interrupt vector table is generally set before RTOS starts, such as in the startup.s file, or when performing an upgrade jump.

If the above code operation is intended to set the following variable '_tx_thread_system_stack_ptr'. It is recommended to let '_tx_thread_system_stack_ptr' be set by the user.

Also the variable '_tx_thread_system_stack_ptr' is not used to reset the interrupt stack.

fdesbiens, 2025-02-27:
My apologies for the long delay. I will flag this issue for discussion with the group of committers.

fdesbiens, 2025-04-01:
@eclipse-threadx/iot-threadx-committers  and @eclipse-threadx/iot-threadx-contributors: please have a look.

cypherbridge, 2025-04-01:
not sure if this is a bug, feature, or preference
my experience is VTOR set here is not a problem

fdesbiens, 2025-04-08:
We discussed this issue and agree the change should be made. 

We will include it in our planning.

cypherbridge, 2026-02-09:
On further review I agree with the recommendation that VTOR and stack vector assigned in ThreadX init should be removed.  VTOR initialization is a function of low level startup which executes before RTOS ThreadX initialization.   Impact to existing code could be a result of this change. Users are advised to regression test and if needed update their startup code adding VTOR and stack.  Typical applications may be able to use the default VTOR initialized by chip reset.

fdesbiens, 2026-09-09:
Thanks for the report, and thanks to @cypherbridge for the review.

You were right on both counts. Setting VTOR inside `_tx_initialize_low_level` is wrong: vector table placement is a function of low-level startup that runs before the kernel is entered, and doing it again in ThreadX overrides whatever the application, a bootloader or a firmware update had already installed. And `_tx_thread_system_stack_ptr` is declared by the Cortex-M scheduler files but never actually read, so copying the reset stack pointer into it accomplished nothing.

PR #714 removes both blocks from every Cortex-M port — 71 files across `ports/`, `ports_arch/` and `ports_module/`, covering the ac5, ac6, gnu, ghs, iar and keil trees — along with the symbol declarations that became unused. As a side benefit, the kernel no longer requires the application to export a vector table symbol.

Please note this is a behaviour change: if your application depended on ThreadX programming VTOR, you now need to set it in your own startup code. Applications whose vector table sits at the reset default address are unaffected.

fdesbiens, 2026-09-09:
The fix has been merged in #714. The VTOR and system stack pointer setup has been removed from `_tx_initialize_low_level` across all 71 Cortex-M port files (ARMv7-M and ARMv8-M, all toolchains).

Applications that relied on ThreadX to program VTOR must now do so in their own startup code — please regression test and update your startup code if needed.

Thank you for the detailed report, and thanks to @cypherbridge for the review and sign-off.

closed 2026-09-09 by fdesbiens