eclipse-threadx/threadx #371: tx_byte_pool service fragmentation problem

SeeDeer, 2024-03-10:
Test method: 
1. Thread A allocate memory blocks in a 1ms cycle (size takes a random value of 1~1024).  When allocate fails, the thread is suspended; 2. Thread B release memory blocks in a 1~10ms cycle. When all memory blocks are released, Restore thread A.

[image]

As shown in the above test results, 

tx_byte_allocate fail, rt:0x10 size:859 frag:208 available:2068
memory msg, frag:208 available:50184
memory msg, frag:208 available:98276
all memory be release, frag:208 available:102392

All memory blocks have been released, why is the frag field not reduced? Are memory blocks at adjacent addresses not merged?

Looking forward to your reply, thank you.

xray-bit, 2024-03-11:
> Adjacent free memory blocks are merged together during a subsequent allocation search for a large enough free memory block. This process is called defragmentation.

https://learn.microsoft.com/en-us/azure/rtos/threadx/chapter3#memory-byte-pools

amgross, 2024-03-11:
As @xray-bit says, the defragmentation happens upon allocation scheme and not free scheme.

This is because in free time there is no need to waste time to look on other blocks, while in allocation time we anyway go block after block to find one that is big enough so meanwhile we also merge consecutive free blocks till we find big enough block and stop.

SeeDeer, 2024-03-15:
OK, thanks for your reply.

closed 2025-02-27 by fdesbiens