eclipse-threadx/threadx #373: How to include threadx in the project via CMake system?

MaJerle, 2024-03-16:
**Is this request related to a particular hardware platform, SoC, board? Please describe.**
There is a CMake build system supported in the system. 
Typically something like below is to be added in the project folder, that will include the Cortex-M33 version of ThreadX in the project:

```cmake
set(THREADX_ARCH "cortex_m33")
set(THREADX_TOOLCHAIN "gnu")
add_subdirectory("../../lib/threadx" threadx)
target_link_libraries(${CMAKE_PROJECT_NAME} threadx)
```

**Describe why you are asking for this support?**
Problem arises, that ARM GCC will throw errors for `*.s` files compilation, complaining about thumb. CMake `compile_commands.json` does not show any sign of global project compiler flags being added to the threadx files.

So how do we correctly add `-thumb` flags to the threadx compiled files, w/o breaking the current structure?
I'm using latest image of the repository.

I guess we are missing something like:

```cmake
set(THREADX_ARCH "cortex_m33")
set(THREADX_TOOLCHAIN "gnu")
set(THREADX_COMPILE_FLAGS "....")             # <- this line
add_subdirectory("../../lib/threadx" threadx)
target_link_libraries(${CMAKE_PROJECT_NAME} threadx)
```

That will then add the values to the `target_compile_options` of `threadx` module.

MaJerle, 2024-03-17:
Furthermore, when researching why I cannot get pass `tx_semaphore_create` function, I noticed that `sizeof(TX_SEMAPHORE)` in the app wasn't the same as `sizeof(TX_SEMAPHORE)` inside threadx library. One was `28`, another was `32`.

A trick is that file `tx_user.h`, that is in the user application folder, outside threadx library, isn't visible to the compilation unit of threadx and therefore doesn't take into account all user settings.

In the `TX_SEMAPHORE` structure, there is this part:
```c
#ifndef TX_DISABLE_NOTIFY_CALLBACKS

    /* Define the application callback routine used to notify the application when
       the a message is sent to the queue.  */
    VOID                (*tx_queue_send_notify)(struct TX_QUEUE_STRUCT *queue_ptr);
#endif
```
Macro `TX_DISABLE_NOTIFY_CALLBACKS` is normally defined in the `tx_user.h` file, but this one is NOT visible to the compilation unit of threadx, if we use the current ThreadX CMake system.

What's the trick here, to make it happen correctly?

Thanks

xray-bit, 2024-03-21:
Hi, @MaJerle . You can do it just like this:

```cmake
set(THREADX_ARCH cortex_m33)
set(THREADX_TOOLCHAIN gnu)
add_subdirectory(threadx)
target_compile_options(threadx PRIVATE
    -mthumb
    # other compile options
)
target_link_libraries(${EXECUTABLE} threadx)
```

fdesbiens, 2025-02-27:
A solution has been proposed. Closing this issue due to inactivity. Feel free to reopen or create a new one.

closed 2025-02-27 by fdesbiens