eclipse-threadx/threadx #377: Certification Package for Eclipse ThreadX

breaultjean, 2024-03-21:
Hi,

I am using Eclipse ThreadX part of the development of a medical device that will be soon going thru certification process. I am now preparing my premarket study and since this is an OTS to my product it needs to go thru certification unless it has been already tested and certified. We specifically chose ThreadX since it has a 62304 independent certification tested on the Azure Sphere MT3620 microcontroller.

I tried to inquiry Microsoft, but they are not managing Thread X anymore. I tried the emo [at] eclipse.org no no availability as of today.

- How will the Eclipse Foundation manage the Certification Packages for ThreadX? 
- Will it still be costing a 30K$ fee?
- Who do I contact?

Info is here: https://github.com/eclipse-threadx/threadx/wiki/Functional-Safety-Artifacts and https://techcommunity.microsoft.com/t5/internet-of-things-blog/building-iot-solutions-with-rtos/ba-p/2420780

Thank you for the support!

ravurinaresh, 2024-03-24:
ThreadX now will be transferred to Eclipse foundation under MIT license including Safety Certifications as of now acquired and these safety certifications and safety documents will be available for Eclipse foundation members.

breaultjean, 2024-03-26:
Ok, I'll send an enquiry directly to the mailing list. Thanks.

closed 2024-03-26 by breaultjean