eclipse-threadx/threadx #380: Fix CMake minimum require version warning

nate-plxs, 2024-04-01:
The `cmake_minimum_required` specified in the parent CMakeLists.txt file is far enough out of date that it triggers a CMake warning starting in CMake 3.27, see [docs](https://cmake.org/cmake/help/latest/command/cmake_minimum_required.html#policy-settings)

This can be fixed, by at a minimum bumping to `cmake_minimum_required(VERSION 3.5.0 FATAL_ERROR)`.

```
Compatibility with CMake < 3.5 will be removed from a future version of
CMake.
Update the VERSION argument <min> value or use a ...<max> suffix to tell
CMake that the project does not need compatibility with older versions.
```

jonny-svaerd-arm, 2025-01-23:
Quick information update for CMake 3.31: compatibility with CMake versions < 3.10 has been deprecated.
Thanks

fdesbiens, 2025-02-27:
The minimal CMake version has been bumped to 3.13 in ThreadX v6.4.2. Thank you for flagging this issue.

closed 2025-02-27 by fdesbiens