eclipse-threadx/threadx #381: Cortex-M7 : Is _tx_thread_context_save/_tx_thread_context_restore required?

maxkunes, 2024-04-02:
The ThreadX docs under Chapter 3 - Functional Components: ISR Template suggest that context saving and restoring must be done in assembly before executing any ThreadX code (i.e. notifying semaphores) or calling user code.

Is this required on Cortex-M7? Or is this only needed on specific architectures?

rayc345, 2024-05-16:
Not manually, For Armv7-M ISA based CPU like Cortex-M7, caller save registers (r0-r3,r12,LR(r14),PC(r15), xPSR) would be saved into stack by hardware, and r4-r9, r10,r11,SP would be saved by callee functions, save and restore instructions would be generated by compilers, so for C programmers, no manual procedure is required to save/restore contexts.

amgross, 2024-05-16:
Hi @rayc345 
Threadx context save/restore not only saves register, but also manages the option that while the interrupt happened something that requires context switch. For this case stuff needed to be stored on the stack and context restore returns from the interrupt to the scheduler instead of to the original point where the interrupt was called from.
And hence I think the answer to @maxkunes is that indeed it should be called from assembly code.

maxkunes, 2024-05-16:
@amgross Thank you for the reply.

Honestly, I've been working with ThreadX for 3 years now without doing this, and I've yet to see anything that is obviously wrong.
Can you explain what behavior might be seen in theory without these calls? Maybe I have some intermittent bugs due to this that I haven't connected to anything else.

amgross, 2024-05-19:
Hi @maxkunes , 
By without doing it you mean you call it from C code or not calling it at all?

maxkunes, 2024-05-19:
I do not call it at all.

amgross, 2024-05-19:
Surprisingly (for me), it seems like I was mistaken and there is no need for cortex M to use save/restore

https://github.com/eclipse-threadx/threadx/blob/485a02faec6edccef14812ddce6844af1d7d2eef/ports/cortex_m7/ac5/readme_threadx.txt#L161-L173

So @rayc345 is correct

maxkunes, 2024-05-19:
But that seems to depend on the compiler used?

See:
https://github.com/eclipse-threadx/threadx/blob/485a02faec6edccef14812ddce6844af1d7d2eef/ports/cortex_m7/gnu/readme_threadx.txt

Which says on GNU you have to use assembly to save r0 and lr and then you
can bl to your C function.

Im using arm-none-eabi-gcc, which presumably is AC5 or 6? If so, that
explains why I don't have issues.

fdesbiens, 2025-02-27:
Closing this issue due to inactivity. Feel free to reopen or create a new one.

closed 2025-02-27 by fdesbiens