eclipse-threadx/threadx #383: Consider declaring the name of the entities as "CHAR const *" instead of "CHAR *"

pirgia, 2024-04-29:
The functions that create an entity (thread, semaphore, etc.), take the entity's name as CHAR*.
I believe that, as the name isn't likely to change, the constness of the name (in the structures and the parameters) would be a nice to have feature.
But the reason of the request is that, while it is allowed in "C" to pass a constant string to a "char *" parameter, in C++ (starting from C++11) it is not allowed to pass a string literal (constant) to a "char *" parameter (ISO C++ forbids converting a string constant to 'CHAR*').
The latter forces the user (interfacing a C++ application with ThreadX) to const-cast the name of the entity, something like:

tx_thread_create(&thread1, const_cast<char*>("Thread name"),....);

which is ugly and, above all, not recommended by many coding standards.
I'd like to know your opinion.
Thanks for your attention.

MaJerle, 2024-04-29:
Fully support this. It should have been done from the day 1.

hwmaier, 2024-10-02:
Thankfully we have now a PR to address this long standing issue.  Refer to https://github.com/eclipse-threadx/threadx/pull/414

fdesbiens, 2025-02-27:
I requested a review on the PR, @hwmaier. If it is positive, this feature will likely be part of ThreadX v6.4.3.

fdesbiens, 2026-09-09:
Thank you for the suggestion, @pirgia, and apologies for the long delay in getting back to you.

You are right on both counts: the name pointers should be `const`, and the current signatures force C++ users into `const_cast` gymnastics that most coding standards discourage.

As @hwmaier pointed out, this is the same request as #61, which has been open since 2020 and is where the design discussion now lives. PR #414 attempted the change but was closed earlier this year after it drifted too far from `dev` to be rebased cleanly.

The work is planned and scoped: the change touches the object control blocks, every `*_create` and `*_name_get` prototype, the module manager interface, and the documentation. The only source-level incompatibility is `tx_*_name_get`, whose out-parameter has to become `const CHAR **`; object creation with string literals keeps working unchanged. I have put #61 on the agenda for the project meeting so the committers can settle the compatibility question, and I am targeting the December release for the implementation.

I am closing this one as a duplicate so the discussion stays in a single place. Please subscribe to #61 to follow along.

closed 2026-09-09 by fdesbiens