eclipse-threadx/threadx #385: ThreadX Context Save/Restore of Floating Point Registers for Cortex-R5

austinawolf, 2024-05-16:
I'm working on setting up ThreadX with Cortex-R5F on Xilinx Ultrascale+ R5F. I've noticed some data corruption associated with using floating point registers on memcpy and now realizing that [ _tx_thread_context_save](https://github.com/eclipse-threadx/threadx/blob/master/ports/cortex_r5/gnu/src/tx_thread_context_save.S#L80) for R5 port doesn't save floating point registers. Doesn't this mean that this port doesn't hard floating point math at all? Is there any reference code to update this logic to support saving floating point registers?

austinawolf, 2024-05-21:
Some of the other ARM ports seem to have support for using `TX_ENABLE_VFP_SUPPORT` then using `tx_thread_vfp_enable` to enable it, but the R5 is missing is the extension define here:
https://github.com/eclipse-threadx/threadx/blob/master/ports/cortex_r5/gnu/inc/tx_port.h#L197

Like for the A9:
https://github.com/eclipse-threadx/threadx/blob/master/ports/cortex_a9/gnu/inc/tx_port.h#L199

ted0xff, 2024-05-30:
Funny, seems related to our #382 two weeks  before your report, we are on the same problematics

Our Quick and Dirty solution is , for now, building in softFP...

billlamiework, 2025-03-19:
It looks like all this is missing are the prototypes in tx_port.h, e.g., the VFP code and enable/disable APIs are in the assembly port files.  You do need to build the assembly code with TX_ENABLE_VFP_SUPPORT defined though.

yuxinzhou5, 2025-04-08:
@austinawolf Following suggestions from @billlamiework:
(1) Add "#define TX_THREAD_EXTENSION_2                  ULONG       tx_thread_vfp_enable; " to your tx_port.h 
(2) Rebuild the ThreadX library and all the application code with the symbol   TX_ENABLE_VFP_SUPPORT defined.  

The  ThreadX team will review a fix for the next release.

austinawolf, 2025-04-09:
Got it, thanks @billlamiework @yuxinzhou5

fdesbiens, 2026-09-10:
Hello @austinawolf, and apologies for the long delay in closing this one out.

Yes, hard floating point works on Cortex-R5 with ThreadX, and the missing piece you identified is exactly right. The VFP context is saved and restored lazily: `tx_thread_context_save.S` deliberately does not touch the floating-point registers, and the save/restore happens instead in `tx_thread_schedule.S`, `tx_thread_system_return.S` and `tx_thread_context_restore.S`, guarded by `TX_ENABLE_VFP_SUPPORT` and keyed off the per-thread `tx_thread_vfp_enable` flag. That flag lives in `TX_THREAD_EXTENSION_2`, which was indeed absent from the Cortex-R5 `tx_port.h` files.

That has since been fixed. The GNU and AC6 ports were corrected in commit `2baae7c5` ("Fixed VFP issues on Cortex-R4 and R5 ports", #578), and the AC5 port was corrected last week in commit `7f6e4962` (#715), which was the last Cortex-R5 toolchain still affected. All five Cortex-R5 ports (ac5, ac6, ghs, gnu, iar) now define `TX_THREAD_EXTENSION_2` as `ULONG tx_thread_vfp_enable;` unconditionally. It has to be unconditional because the assembly reads the field at the hard-coded displacement `[thread, #144]`; making it conditional would shift every following member and be correct in only one build configuration.

To use hard float on the R5, build both the ThreadX library and your entire application with `TX_ENABLE_VFP_SUPPORT` defined, then call `tx_thread_vfp_enable()` from each thread that uses floating point. Threads that never enable it pay no context-switch cost. Note that the whole application must be built with the same setting, since the flag changes the layout of `TX_THREAD`, so a partial rebuild will produce exactly the kind of silent data corruption you were seeing.

I am closing this as completed. Please reopen or file a new issue if you still see corruption on the current `dev` branch.

closed 2026-09-10 by fdesbiens