eclipse-threadx/threadx #386: Consulting

fliiiix, 2024-05-17:
We struggle a lot with the performance of the GNU/posix port at work and I was looking for consulting in that area. 

I think it would be great to have a list of people which sell support and consulting for ThreadX.

(I'm not really sure where this request fits best, in case it's not this repo im sorry)

Hnz2, 2024-05-17:
Hi,

If you are interesting to buy support for ThreadX, you can ask [here](https://rtosx.com/).

Jan

fliiiix, 2024-05-27:
I already tried the rtosx people :sweat_smile: , but they did not get back to me so if anyone has any other leads i would be very interested

MaJerle, 2024-05-27:
@billlamiework

williamelamie, 2024-05-27:
We (RTOSX) haven't seen your request. Please send an e-mail to support@rtosx.com or me directly at blamie@px5rtos.com

rahmanih, 2024-06-27:
closing as answered 😉