eclipse-threadx/threadx #389: FPU support RISCV-64

Alexbruliau, 2024-05-29:
**Describe the bug**
FPU is deactivated inside tx_thread_schedule.S v6.4.1

[image]

Is it the correct behavior ? 
I would expect the FS bit from the mstatus not to be reset to 0. Otherwise it will cause a trap if you try to access a FP register afterwards.

xuzihan351, 2024-05-30:
Oops, I had modified there as follow, and it work well:
#ifdef __riscv_flen
    li      t0, 0x3880                                  # Prepare MPIP
#else
    li      t0, 0x1880                                  # Prepare MPIP
#endif

fdesbiens, 2025-02-27:
Closing this issue due to inactivity. Feel free to reopen.

closed 2025-02-27 by fdesbiens