eclipse-threadx/threadx #390: Release Cadence

alberk8, 2024-05-30:
Since Eclipse foundation took over this project, I don't see a whole lot of commits or development.  Is there a roadmap or release cadence as to what is planned?

alberk8, 2024-06-12:
Thank you for the update.

xray-bit, 2024-07-10:
any progress?

fdesbiens, 2024-12-17:
Hi everyone.

I appreciate your patience. The project is in much better shape now. We started merging small fixes and improvements and are working on a public roadmap, which we intend to make available in early 2025. 

I expect we will do a v6.4.2 service release in Q1 2025, barring any unforeseen issues. 

My current focus is to convert the current documentation set (rtos-docs repository) from markdown to AsciiDoc. This is because we found a way to publish the current MD files to HTML but ran into difficulties in doing the same to PDF. Microsoft used a custom solution for that, and we did not inherit it. Moving to AsciiDoc will enable us to publish easily to both formats.

So, things are looking up! Happy holidays to you all.

tdjastrzebski, 2024-12-17:
I am glad to hear that, especially since in my opinion ThreadX is a piece of very good software. Probably its major problem was that it was seen as a Microsoft product.

fdesbiens, 2025-02-27:
We released ThreadX v6.4.2 last week.

See the details in this email I sent: https://www.eclipse.org/lists/threadx/msg00127.html

We will soon start planning our future releases in a public backlog. Stay tuned!