eclipse-threadx/threadx #393: Will ThreadX support osDelayUntil function in the future？

IACU, 2024-06-18:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Hnz2, 2024-06-18:
Hi IACU,

But if this is important for you, you can calculate relative delay from _tx_time_get()_. I don't see any reason to have implemented API like _osDelayUntil()_.

Jan

Hnz2, 2024-06-18:
Hi @IACU,
I don't believe that RTOS sleep at any type of RTOS was designed for such precision timing. Even with FreeRTOS you will see jitter near to RTOS tick when _osDelayUntil()_ API is used. For more precisions timing should be used another approaches (e.g. hardware timers with semaphore, events or message queue).
Jan

IACU, 2024-06-18:
> Hi @IACU, I don't believe that RTOS sleep at any type of RTOS was designed for such precision timing. Even with FreeRTOS you will see jitter near to RTOS tick when _osDelayUntil()_ API is used. For [...]

OK, thanks for your reply

rahmanih, 2024-06-27:
closing this item as answered.

closed 2024-06-27 by rahmanih