eclipse-threadx/threadx #396: Error running sample demo

z3r0g2017, 2024-06-30:
Hi All,

I was able to build threadx as per README as below -

```
vscode@f5db10ae23ff:/workspaces/threadx-master$ cmake -Bbuild -GNinja -DCMAKE_TOOLCHAIN_FILE=../cmake/cortex_m4.cmake .
-- The C compiler identification is GNU 13.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- THREADX_ARCH: cortex_m4
-- THREADX_TOOLCHAIN: gnu
-- Using default tx_user.h file
got here
-- Configuring done
-- Generating done
-- Build files have been written to: /workspaces/threadx-master/build
vscode@f5db10ae23ff:/workspaces/threadx-master$ cmake --build ./build
[193/193] Linking C static library libthreadx.a
vscode@f5db10ae23ff:/workspaces/threadx-master$ echo $?
0
vscode@f5db10ae23ff:/workspaces/threadx-master$ ls ./build/libthreadx.a 
./build/libthreadx.a
vscode@f5db10ae23ff:/workspaces/threadx-master$ 

```
But, when I try to run the sample threadx demo in a devcontainer from vscode, I get an error. I created a CMakeLists.txt as below. 
```
cmake_minimum_required(VERSION 3.0.0 FATAL_ERROR)
project(demo)

# set(THREADX_ARCH linux)
set(THREADX_ARCH cortex_m4)
set(THREADX_TOOLCHAIN gnu)

add_subdirectory(.. threadx)
# add_library(threadx)

target_compile_options(threadx PRIVATE
    -mthumb
    # other compile options
)

add_executable(${PROJECT_NAME} demo_threadx.c)
target_link_libraries(${EXECUTABLE} threadx)
include_directories(SYSTEM BEFORE ../common/inc ../ports/linux/gnu/inc /usr/include ../build)
```
The error I am getting is -

```
vscode@f5db10ae23ff:/workspaces/threadx-master/samples$ cmake -Bbuild -GNinja -DCMAKE_TOOLCHAIN_FILE=../cmake/cortex_m4.cmake .
-- The C compiler identification is GNU 13.2.1
-- The CXX compiler identification is GNU 13.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc
-- THREADX_ARCH: cortex_m4
-- THREADX_TOOLCHAIN: gnu
-- Using default tx_user.h file
got here
-- Configuring done
-- Generating done
-- Build files have been written to: /workspaces/threadx-master/samples/build
vscode@f5db10ae23ff:/workspaces/threadx-master/samples$ cmake --build ./build
[28/195] Linking C executable demo
FAILED: demo 
: && /opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc -mthumb -mcpu=cortex-m4  --specs=nosys.specs -fdata-sections -ffunction-sections -mlong-calls -mthumb -mcpu=cortex-m4  -Wl,--gc-sections CMakeFiles/demo.dir/demo_threadx.c.obj -o demo   && :
/opt/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /opt/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-closer.o): in function `_close_r':
closer.c:(.text._close_r+0xc): warning: _close is not implemented and will always fail
/opt/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /opt/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-lseekr.o): in function `_lseek_r':
lseekr.c:(.text._lseek_r+0x14): warning: _lseek is not implemented and will always fail
/opt/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /opt/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-readr.o): in function `_read_r':
readr.c:(.text._read_r+0x14): warning: _read is not implemented and will always fail
/opt/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /opt/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-writer.o): in function `_write_r':
writer.c:(.text._write_r+0x14): warning: _write is not implemented and will always fail
/opt/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: CMakeFiles/demo.dir/demo_threadx.c.obj: in function `main':
demo_threadx.c:(.text.main+0x10): undefined reference to `_tx_initialize_kernel_enter'
collect2: error: ld returned 1 exit status
[45/195] Building C object threadx/CMakeFiles/threadx.dir/common/src/tx_initialize_kernel_setup.c.obj
ninja: build stopped: subcommand failed.
vscode@f5db10ae23ff:/workspaces/threadx-master/samples$ 

```
Can someone help me troubleshoot and fix this ?
TIA

xray-bit, 2024-08-23:
Your GCC does not implement these functions. You need to achieve it yourself. This is not a problem with Threadx itself.

fdesbiens, 2025-02-27:
My apologies for the delayed response, @z3r0g2017. I assume that you either gave up or found a solution by yourself. I will close the issue. Please create a new one if you need assistance.

closed 2025-02-27 by fdesbiens