eclipse-threadx/threadx #397: Nested comments causes compiler warnings

jltrahan-Gecko, 2024-07-04:
**Describe the bug**
tx_thread_context_save.s in the rxv3 port contains a nested comment, which causes a compiler warning (gcc for RX in this case, v8.3.0-202405.)

Specifically, line 97 contains a nested "C-style" comment that started at line 88.  I'm not sure if this is a ThreadX issue, or specifically gcc's (i.e. gcc ASM should be ignoring the *entire line* to begin with, as it starts with a ;

Other compilers/IDEs (CC-RX, IAR) may have this issue, I have not checked.

**To Reproduce**
Simply build an rxv3 project (with the relevant port) with gcc for RX.

**Expected behavior**
No warning - i.e. remove next comment

**Impact**
Annoyance, need to "filter" out non-critical warning.

jltrahan-Gecko, 2024-07-05:
Hi,

I also opened a support ticket with the maintainers of gcc for RX (https://llvm-gcc-renesas.com/)

They responded that .s files are sent directly to the assembler, but that .S files (which is the case here) are parsed by the C preprocessor first, _before_ being sent to the assembler.  I tested, and confirmed this behavior.

Since using the C preprocessor may be useful, even with assembly files, I think the .S extension should stay.  This means that the underlying issue, the unclosed comment, should be resolved (i.e. add closing "*/" at the end of line 95 of tx_thread_context_save.S)

fdesbiens, 2025-02-27:
Anyone in @eclipse-threadx/iot-threadx-contributors willing to tackle this issue?

fdesbiens, 2025-03-19:
@eclipse-threadx/iot-threadx-committers and @eclipse-threadx/iot-threadx-contributors: Please  provide your feedback.

sanderson-AstroFirmwareSolutions, 2025-04-22:
I agree with @jltrahan-Gecko, the nested comment should be closed by adding "*/" at the end of line 95. Some other ports, such as rxv2 and rxv1, also have this issue. I'll work on finding instances of nested comments and create a PR.

jltrahan-Gecko, 2025-05-21:
Any progress on this?

jltrahan-Gecko, 2025-06-23:
@sanderson-AstroFirmwareSolutions @fdesbiens Hi guys, is this issue on anyone's radar?

fdesbiens, 2025-06-24:
Hi @jltrahan-Gecko.

 Yes, this is on our radar. We are a <small team and are currently working on higher priority tasks right now. I hope to fix this in our next release, nevertheless.

@sanderson-AstroFirmwareSolutions: Do you still intend to submit a PR for this issue?

fdesbiens, 2026-09-23:
I confirmed that #550 fixed this warning on `dev` for the RXv1, RXv2, and RXv3 GNU ports. I also re-ran preprocessing with `-Wcomment -Werror=comment` for all three affected files; all completed without a diagnostic. I am closing this as fixed.

closed 2026-09-23 by fdesbiens

francdoc, 2026-09-23:
Hi [[@fdesbiens](https://github.com/fdesbiens)](https://github.com/fdesbiens), thank you for taking the time to review and merge [[PR #550](https://github.com/eclipse-threadx/threadx/pull/550)](https://github.com/eclipse-threadx/threadx/pull/550).