eclipse-threadx/threadx #398: threadx support stm32f051?

darcyliu2008, 2024-07-08:
we dont't find cortex_m0 folder in ports_module; did threadx support cortex_m0; and how can we do?

darcyliu2008, 2024-07-08:
for gun

darcyliu2008, 2024-07-08:
for gnu

xray-bit, 2024-07-09:
threadx supports m0

https://github.com/eclipse-threadx/threadx/tree/master/ports/cortex_m0

but dynamic module not support m0

darcyliu2008, 2024-07-11:
thanks!

yes, i can't find supports in 
https://github.com/eclipse-threadx/threadx/tree/master/ports_module

if i need to support, how can i do?

darcyliu2008, 2024-07-17:
tx_misra.s assembly file for the port for the M0 kernel  not find, but i find in m4 [tx_misra.S](https://github.com/eclipse-threadx/threadx/blob/master/ports/cortex_m4/gnu/src/tx_misra.S);

darcyliu2008, 2024-07-17:
if tx_misra.s not need in M0?

fdesbiens, 2025-02-27:
Closing this issue due to inactivity. Feel free to reopen if you still need assistance.

closed 2025-02-27 by fdesbiens