eclipse-threadx/threadx #400: Threadx modules kernel stack memory leakage

amgross, 2024-07-29:
**Describe the bug**
tx_thread_module_kernel_stack_start is allocated at _txm_module_manager_thread_create from the object pool, but is not released at thread delete function (it is deleted upon module stop when all objects of the module are deleted)

**To Reproduce**
```
    while(1)
    {
		status = txm_module_object_allocate(<thread_ptr>, sizeof(TX_THREAD));
		assert(TX_SUCCESS == status);

		status = tx_thread_create(<thread_ptr>, "my thread", <my_thread_func>, 0, <stack_ptr>,
                <stack size>, <priority>, <priority>, <time slice>, TX_DONT_START);
		assert(TX_SUCCESS == status);

		status = tx_thread_terminate(<thread_ptr>);
		assert(TX_SUCCESS == status);

		tx_thread_delete(<thread_ptr>);
		assert(TX_SUCCESS == status);
    }
```

**Expected behavior**
tx_thread_module_kernel_stack_start should be released upon thread delete.

**Impact**
Can leak object pool out of memory, which cause no more option to create OS objects from module

**Logs and console output**

**Additional context**

amgross, 2024-07-29:
Workaround: add to tx_port.h (or tx_user.h)
```
extern UINT  _txm_module_manager_object_deallocate(VOID *object_ptr);
#define TX_THREAD_DELETE_PORT_COMPLETION(t) _txm_module_manager_object_deallocate(t -> tx_thread_module_kernel_stack_start);
```

fdesbiens, 2025-02-27:
Anyone in @eclipse-threadx/iot-threadx-contributors willing to tackle this one?

fdesbiens, 2025-03-19:
@eclipse-threadx/iot-threadx-committers and @eclipse-threadx/iot-threadx-contributors: Please  provide your feedback.

yuxinzhou5, 2025-04-08:
Agree with @amgross' workaround for now while we figure out a proper fix.

sravani-25-03-2003, 2025-04-19:
please provide the  SNMP configurations with Azzure rtos (netxduo) im confusing about configurations , please help me anybody

DrBinary9, 2025-09-16:
@fdesbiens @amgross @yuxinzhou5 @sravani-25-03-2003 I would like to take this issue up. Can anyone please point out where these changes need to be made?

prashit-vora, 2026-08-27:
Hi @fdesbiens , is anyone actively working on this? If not, I would love to take it up.

fdesbiens, 2026-08-27:
Hi @prashit-vora.

No one is currently working on it. Feel free to pick it up.

prashit-vora, 2026-09-03:
I opened #692 with a fix for this.

I also reproduced the leak using a temporary module test running under QEMU. The test repeatedly allocated, created, terminated, and deleted a user-mode thread.

Without the fix, thread creation failed with `TX_NO_MEMORY` after 18 iterations. With the fix, all 64 iterations completed successfully.

The QEMU test harness is not included in the PR. I can add it as a permanent regression test if that would be useful.

fdesbiens, 2026-09-08:
Fixed by #692, now merged to `dev`. Thanks for the report, @amgross, and thanks to @prashit-vora for the fix and the QEMU reproduction.

The root cause was exactly as you described. `_txm_module_manager_thread_create()` allocates a separate syscall kernel stack from the module object pool whenever the module carries the `TXM_MODULE_USER_MODE` property, but the `tx_thread_delete()` dispatcher only released the `TX_THREAD` object and left that kernel stack allocated. The memory was reclaimed only when the module was stopped, since `_txm_module_manager_stop()` drains the entire allocated-object list, so a module that repeatedly created and deleted threads at run time would exhaust its object pool.

The dispatcher now releases the kernel stack once `_txe_thread_delete()` succeeds, and it always releases the thread object afterwards so that a kernel stack failure can never leave the thread object stranded.

Closing this one. Please reopen if you still see the pool shrink in your application.

closed 2026-09-08 by fdesbiens