eclipse-threadx/threadx #401: Define system clock as ULONG64

joelguittet, 2024-08-08:
**Is your feature request related to a problem? Please describe.**
Currently the system clock `_tx_timer_system_clock` is defined as an ULONG variable, so it's 32-bits unsigned, maximum value is 4294967295. With standard configuration using a 10ms timer tick, this means the timer will overlap avec 497 days only.
While it's maybe not a problem, it can be very nice to set the system time with the current epoch time.
That means currently taking the epoch time and multiply by `TX_TIMER_TICKS_PER_SECOND` to call `tx_time_set` function
Problem is actually with such resolution this overlap the ULONG internal system time.
Example: Thursday 8 August 2024 20:00:00 GMT is 1723147200 so multiply by 100 this means 172314720000 which is larger than 4294967295.

**Describe the solution you'd like**
I would like the internal system clock `_tx_timer_system_clock` to be defined as an ULONG64 so that there is no overlap at all to consider and we can set the wanted system time with the wanted timer tick resolution. `tx_timer_set` should take an ULONG64 type. `tx_timer_get` should returns an ULONG64 type as well.
I can propose a Pull Request if this is appreciated.

**Describe alternatives you've considered**
Maintaining myself the system time outside of ThreadX. Not very convenient.

**Additional context**
My application has an SNTP client that periodically check for NTP time. It would be great to set have such feature so that the SNTP client just update the ThreadX system time and then other modules of the application just call the ThreadX API to get system time.

fdesbiens, 2025-02-27:
My apologies for the delayed feedback, @joelguittet.

I will flag this issue for discussion with the committer team. I should be able to get back to you in the upcoming weeks.

fdesbiens, 2025-03-19:
@eclipse-threadx/iot-threadx-committers and @eclipse-threadx/iot-threadx-contributors: Please  provide your feedback.

billlamiework, 2025-03-19:
Unfortunately, this API and functionality change could impact existing customers. The only option would be to make this configurable under conditional compilation, where the current behavior is the default.

joelguittet, 2025-03-20:
@billlamiework sounds a good option to me. If I make a PR, would it be reviewed ?

fdesbiens, 2025-04-01:
@joelguittet Yes, please submit a PR.

tdjastrzebski, 2025-08-10:
very good idea, long awaited feature

fdesbiens, 2026-09-23:
I appreciate the proposal and the discussion here. I have opened #767 to track this as a new feature request with current compatibility, portability, testing, and documentation requirements. I am closing this issue in favor of #767.

I want to be clear that I have not assigned any maintainer or contributor to implement this feature, and I do not have a delivery schedule for it. I welcome contributions that satisfy the requirements in the new issue.

closed 2026-09-23 by fdesbiens