eclipse-threadx/threadx #408: make threadx compiles with c++ compilers

rahmanih, 2024-09-05:
Hi,
When compiling threadx with g++ compilers, some errors are thrown.
this request is about fixing all those errors.

PS: attached the error log when compilinga project using  arm-none-g++.(not all files where compiled in this project).
[build_errors.txt](https://github.com/user-attachments/files/16897089/build_errors.txt)

hwmaier, 2024-09-06:
It appears you are using `g++` to compile ThreadX code. ThreadX is written in C not C++, so try using `gcc` instead of `g++`.

In case you want to compile C++ code with ThreadX header files, set the compiler flag `-fpermissive`.

I tend to add the following line to my CMakefiles for mixed language (C/C++)projects:
```
add_compile_options($<$<COMPILE_LANGUAGE:CXX>:-fpermissive>) # Allow passing const char * to char * in C++
```

I hope that helps to overcome your compilation issue.

fdesbiens, 2025-02-27:
Closing this issue due to inactivity. Feel free to reopen it if you still need support, @rahmanih. Thank you for helping out, @hwmaier.

closed 2025-02-27 by fdesbiens