eclipse-threadx/threadx #410: Is this project dead?

xray-bit, 2024-09-26:


Hnz2, 2024-09-26:
I afraid that it is.

amgross, 2024-09-26:
From what I understand no, it is just in freeze for while due to long transition of it from microsoft to eclipse.
You can connect to the eclipse threadx mailing list

fdesbiens, 2024-09-26:
@amgross is right, @Hnz2.

We have been focused on putting things into place what we need to provide access to the safety manuals and support the certification of ThreadX 6.4.1. 

We fixed some CVEs a few months ago. 

Some of the committers have been working on changes which are not done yet.

The mailing list is here: https://accounts.eclipse.org/mailing-list/threadx

You need an Eclipse Foundation account to subscribe.

More to come. We have an announcement coming.

xray-bit, 2024-09-27:
okay. thanks for letting us know.

closed 2024-09-27 by xray-bit