eclipse-threadx/threadx #412: SMP trace status

yf13, 2024-09-30:
I am debugging a SMP port here and noticed that when trace is disabled the sample works more stably. Thus I am wondering if the trace function within SMP ThreadX needs special care when using?

I am also wondering if Trace data has considerations for SMP use cases? It seems that  the highest byte in event id is core id, but not sure if things like core standy and wakeup are captured?

yf13, 2024-10-14:
> It would be good to understand better the strange issues that occur when trace is enabled and used, e.g., are the issues related to the operation of ThreadX or problems with the trace buffer?

@billamiework, thanks for asking. I often saw message receiving error together with instruction access faults on QEMU RiscV v6.2. I rotated two trace buffers upon buffer full notification.

My sample quits when `thread_0_counter` reaches 1000 (with 1 tick sleep), terminates other app threads, print out statistics and quit QEMU.

So a normal console output looks like below, with 97 rotates of trace buffers:

```
Copyright (c) 2024 Eclipse Foundation. * ThreadX SMP/RISC-V/GNU Version 6.4.x *
**** ThreadX RiscV Demonstration **** (c) 1996-2024 Eclipse Foundation

           thread 0 events sent:          1000
           thread 1 messages sent:        500
           thread 2 messages received:    501
           thread 3 obtained semaphore:   251
           thread 4 obtained semaphore:   251
           thread 5 events received:      1001
           thread 6 mutex obtained:       251
           thread 7 mutex obtained:       251
           thread 8 messages received:    97

Stack depths: 824, 776, 764, 692, 776, 884, 816, 820, 484, 644
SMP inter-core interrupts: 641, 1006
```

Below is for typical failure case:

```
Thread_2.0 error: got 0.525322, expect -0.432178(46)

Instruction access fault(1) #0 epc: 0 tva: 0 mstatus: 1880 lvl:1 thr:thread 2, cnt:94 sp: 80081c20(800816ac 80081ca7) ra:0
```

If trace have been used with other SMP ports before then it is likely an issue specific to my port.

yf13, 2024-10-14:
> It would be good to understand better the strange issues that occur when trace is enabled and used, e.g., are the issues related to the operation of ThreadX or problems with the trace buffer?

@billamiework, thanks for asking. I think it more relates to ThreadX operations of my port.

My sample rotates two trace buffers upon buffer full and a thread_8 is added to record full buffer to files. The sample quits when `thread_0_counter` reaches 1000 (with 1 tick sleep), terminates other app threads, print out statistics and quit QEMU.

So console log of a successful run looks like below, with 97 trace buffer rotates:

```
Copyright (c) 2024 Eclipse Foundation. * ThreadX SMP/RISC-V/GNU Version 6.4.x *
**** ThreadX RiscV Demonstration **** (c) 1996-2024 Eclipse Foundation

           thread 0 events sent:          1000
           thread 1 messages sent:        500
           thread 2 messages received:    501
           thread 3 obtained semaphore:   251
           thread 4 obtained semaphore:   251
           thread 5 events received:      1001
           thread 6 mutex obtained:       251
           thread 7 mutex obtained:       251
           thread 8 messages received:    97

Stack depths: 824, 776, 764, 692, 776, 884, 816, 820, 484, 644
SMP inter-core interrupts: 641, 1006
```

Below is that for typical failure case with the weird exception:

```
Thread_2.0 error: got 0.525322, expect -0.432178(46)

Instruction access fault(1) #0 epc: 0 tva: 0 mstatus: 1880 lvl:1 thr:thread 2, cnt:94 sp: 80081c20(800816ac 80081ca7) ra:0
```

I tried skipping the data recording, but the issues still occur, so it seems not like a trace buffer contention issue.

If trace have been used with other SMP ports before then it is likely an issue specific to my port.

yf13, 2024-10-15:
@billamiework, I noticed that trace feature has been used with other upstream SMP ports but they don't rotate trace buffers.

So I commented out buffer full notifier registration in my sample and it works as good as trace disabled situation.

So it seems that buffer rotation might be more suspicious:

```c
/* Trace buffer full handler, called within critical section */
void trace_buffer_rotate(void *buffer)
{
void *new;

    tx_trace_disable();

    /* Switch buffer */
    if (buffer == &trace_buffer0) {
        new = &trace_buffer1;
    } else {
        new = &trace_buffer0;
    }
    trace_buffer_setup(new);

    /* Send messag to notify the logging thread  */
    tx_queue_send(&queue_1, &buffer, TX_NO_WAIT);
}
```

Is this the right use of buffer rotation?

billlamiework, 2024-10-15:
It is good that you have narrowed it down to this callback handler…  It would be good to check the thread or ISR stack availability where this callback occurs.  Memory corruption from stack overflow is an easy way to cause a system failure. It might be interesting to know if there is any consistency on where the callback is being called from when the crash occurs. Does it work sometimes, or does it crash every time this is called?  Also, what is going on inside of “trace_buffer_setup”?

billlamiework, 2024-10-16:
> @billlamiework<https://github.com/billlamiework>, thanks again for continuing this.

The trace_buffer_setup processing is non-trivial and, as implemented, could be called from ISR processing. Instead, I suggest using the buffer full callback notification to wake up a thread, which will then perform the trace buffer swap. I don’t know if this is the actual problem…  but it’s hard to tell without digging in with the debugger.

For rv32, stacks for app and timers threads are of 1536 bytes, as per the logs from program finish, they look fine so far. The system stacks are of 2048 bytes, I need see how to track the system stack usage, I might need add some code to track its depth for sure.

For rv64, stacks for app and timer threads are 2048 bytes, the system stacks are of 4096 bytes. here is one log of rv64 success case:

           thread 1 messages sent:        503

           thread 2 messages received:    504

           thread 3 obtained semaphore:   253

           thread 4 obtained semaphore:   252

Stack depths: 1048, 1184, 1184, 1184, 1184, 1172, 1192, 1204, 768, 1096

SMP inter-core interrupts: 1424, 554

The log shows that exception happened to different app threads. Is there a simple way to find where the callback is called?

The demo works sometimes, but often fails. Typical failure behaviours are two: one is above mentioned exception, the other is stalled QEMU which I have to quit forcedly.

Here it is:

void trace_buffer_setup(void *buffer)

{

UINT ret;

    ret = tx_trace_enable(buffer, DEMO_TRACE_SIZE, DEMO_TRACE_OBJECTS);

    if (ret != TX_SUCCESS){

        printf("tx_trace_enable failed: %d", ret);

        exit(1);

    }

    ret = tx_trace_buffer_full_notify(trace_buffer_rotate);

    if (ret != TX_SUCCESS) {

        printf("failed to install buffer full handler: %d", ret);

        exit(2);

    }

}

I haven't find an API to just set the event buffer and keep reusing existing header and object registry, so this is what I have now.

yf13, 2024-10-16:
> Instead, I suggest using the buffer full callback notification to wake up a thread, which will then perform the trace buffer swap.

Thank you @billlamiework, I guess you mean to keep `tx_queue_send()` only in the callback? Will give this a try later as I am using a bigger buffer currently.

As a side topic, I am wondering if we shall enrich trace design later so that more SMP related events are supported?

yf13, 2024-10-17:
@billamiework, I tried following buffer full callback and off-load rotation to a thread:

```c
/* The buffer full callback */
void trace_buffer_full(void *buffer)
{
    tx_trace_disable();
    tx_queue_send(&queue_1, &buffer, TX_NO_WAIT);
}

/* Trace rotation for thread #8 */
void trace_buffer_rotate(void *buffer)
{
TX_INTERRUPT_SAVE_AREA
void *new;

    TX_DISABLE
    if (buffer == &trace_buffer0) {
        new = &trace_buffer1;
    } else {
        new = &trace_buffer0;
    }
    trace_buffer_setup(new);
    TX_RESTORE
}

/* Enable trace with given buffer */
void trace_buffer_setup(void *buffer)
{
UINT ret;
    ret = tx_trace_enable(buffer, DEMO_TRACE_SIZE, DEMO_TRACE_OBJECTS);
    if (ret != TX_SUCCESS){
        printf("tx_trace_enable failed: %d", ret);
        exit(1);
    }
    ret = tx_trace_buffer_full_notify(trace_buffer_full);
    if (ret != TX_SUCCESS) {
        printf("failed to install trace buffer fully handler: %d", ret);
        exit(2);
    }
}

```

Unfortunately it has doesn't help, there are more frequent stalls and errors.

I am wondering how trace buffer rotation works on other SMP ports? I know little about cortex-a SMP ports, thus can't check it on my own yet.

fdesbiens, 2025-02-27:
Do you still need assistance, @yf13? I will close the issue otherwise.

fdesbiens, 2026-09-23:
Eclipse ThreadX does not provide a RISC-V SMP port for now, although this is on our roadmap. The upstream RISC-V 32-bit and 64-bit ports use the uniprocessor kernel and configure QEMU with one core, so I understand this report to concern your own out-of-tree SMP port.

I tested the shared SMP trace implementation on the supported Linux/GNU SMP port with producer threads pinned across all four cores. I ran direct callback rotation, deferred worker-thread rotation, and deferred rotation with the additional `TX_DISABLE` from the sample 21 times each, with 1,000 rotations per run. I observed no stalls, corruption, lost notifications, or API errors across 63,000 rotations.

I cannot reproduce a defect in the shared ThreadX SMP trace code. I also cannot diagnose the RISC-V-specific failure without the complete port source and a minimal sample, including the queue configuration. I am closing this issue, but I can reopen it if those materials become available.

closed 2026-09-23 by fdesbiens