eclipse-threadx/threadx #415: Modules with SMP ports status

yf13, 2024-10-17:
@billamiework, I tried following buffer full callback and off-load rotation to a thread:

```c
/* The buffer full callback */
void trace_buffer_full(void *buffer)
{
    tx_trace_disable();
    tx_queue_send(&queue_1, &buffer, TX_NO_WAIT);
}

/* Trace rotation for thread #8 */
void trace_buffer_rotate(void *buffer)
{
TX_INTERRUPT_SAVE_AREA
void *new;

    TX_DISABLE
    if (buffer == &trace_buffer0) {
        new = &trace_buffer1;
    } else {
        new = &trace_buffer0;
    }
    trace_buffer_setup(new);
    TX_RESTORE
}

/* Enable trace with given buffer */
void trace_buffer_setup(void *buffer)
{
UINT ret;
    ret = tx_trace_enable(buffer, DEMO_TRACE_SIZE, DEMO_TRACE_OBJECTS);
    if (ret != TX_SUCCESS){
        printf("tx_trace_enable failed: %d", ret);
        exit(1);
    }
    ret = tx_trace_buffer_full_notify(trace_buffer_full);
    if (ret != TX_SUCCESS) {
        printf("failed to install trace buffer fully handler: %d", ret);
        exit(2);
    }
}

```

Unfortunately it has doesn't help, there are more frequent stalls and errors.

I am wondering how trace buffer rotation works on other SMP ports? I know little about cortex-a SMP ports, thus can't check it on my own yet.

fdesbiens, 2024-11-21:
Thank you for reaching out @yf13. I will discuss this with the team.

billlamiework, 2024-11-21:
Hi @yf13,

It looks like you are correct - I see the ports_module/cortex_a35_smp folder and the contents do appear to support SMP mode. Don't forget that you should also make sure you are using the threadx/common_smp generic source code (non Module source). For example, there are macros that convert _tx_thread_execute_ptr into an array (indexed by core) for SMP use.  Hopefully, that will helps!

fdesbiens, 2024-12-12:
@yf13 Do you still need assistance? Otherwise, I will close this issue.

yf13, 2024-12-13:
@billlamiework and @fdesbiens, thanks for the information.
Acutally I am still having puzzles. The `txm_module_manager_thread_create.c` belongs to `common_modules/module_manager`. Thus it doesn't support SMP.

If we are not missing a `common_modules_smp/` folder, then the current version of `common_modules/` should be upgraded to support SMP?

billlamiework, 2024-12-13:
Hi @yf13,
Have you actually tried to build an SMP version or is it just concerning that there isn't a common_modules_smp folder?  I ask because there are macro redefines for SMP that convert ThreadX from single core to SMP.

billlamiework, 2024-12-15:
Hi @yf13,
I see the build error now at the beginning of the thread, and it does appear something is missing with SMP and module support. If you want to discuss some options for recreating this support, please feel free to reach out to me via email: blamie@px5rtos.com

fdesbiens, 2025-02-27:
Closing this issue due to inactivity. Feel free to reopen or create a new one as needed.

closed 2025-02-27 by fdesbiens