eclipse-threadx/threadx #417: Download the ThreadX repo, could not get the requirment branch

Huazi715, 2024-11-12:
Hi,
git clone the thread core code.  But there are not the v6.1.3 branch.

How can I get the correct branch 6.1.3? 

[image]
[image]

fdesbiens, 2024-11-12:
Hi @Huazi715.

Releases for ThreadX and its modules are tagged; they do not have their own branch after release.

For your convenience, here are the tags for the latest certified components.

| Component name | Version | GitHub Tag |
| -------------- | ------- | ---------- |
| ThreadX | 6.1.1 | https://github.com/eclipse-threadx/threadx/releases/tag/v6.1.1_rel |
| ThreadX SMP | 6.1.3 | https://github.com/eclipse-threadx/threadx/releases/tag/v6.1.3_rel |
| GUIX | 6.1.7 | https://github.com/eclipse-threadx/guix/releases/tag/v6.1.7_rel |
| NetX Duo | 6.1.9 | https://github.com/eclipse-threadx/netxduo/releases/tag/v6.1.9_rel |
| USBX | 6.1.11 | https://github.com/eclipse-threadx/usbx/releases/tag/v6.1.11_rel |

Please note that ThreadX and ThreadX SMP are in the same GitHub repository, but their certified versions differ.

I hope this helps. Let me know if you have other questions; otherwise, I will close this issue.

Huazi715, 2024-11-12:
Hi Frédéric Desbiens,

Yeah, I know this way, but this download method is the zip or tar.gz format, I could check merge log by the cmd git log.

Is there other download method that can check merge log？

BR/Hua

fdesbiens, 2024-11-12:
Hi Hua.

You can checkout any tag in a branch. For example, if you wanted to checkout ThreadX 6.1.3 (tag `v6.1.3_rel`) and create a local branch named `dev` to work on the code, you would issue the following command:

```
git checkout -b dev tags/v6.1.3_rel
```

Huazi715, 2024-11-13:
Yeah,  got it. Thank you.
I will close this issue.

closed 2024-11-13 by Huazi715