eclipse-threadx/threadx #418: When mutex already acquired by lower priority thread, the higher priority thread also get the same mutex without thread suspend

devaraj-siva, 2024-11-12:
**Description of the bug:**
The mutex was acquired by lower priority thread and performing its thread activity and not yet released the mutex. When higher priority thread is trying to acquire the same mutex, still _tx_mutex_get() function call return TX_SUCCESS. So higher priority thread perform its thread activity and releasing the mutex. When mutex released from higher priority thread, _tx_mutex_put() function returned the value TX_NOT_OWNED. Because the mutex owner is lower priority thread. Due to TX_NOT_OWNED return value, our firmware is designed to generate software assertion.

**Additional information:**
Target - MIPS Interaptiv SMP CPU
Eclipse ThreadX version - threadx-6.4.1_rel
Compiler - GHS Multi compiler

Initially we tested our application in 50 MHz CPU clock. The application is working fine.
When we test our application in 1.1 GHz (1100 MHz) CPU clock, this issue is observed.

We debugged the issue and found the root cause.
**Working scenario:**
When higher priority thread try to get already acquired mutex, higher priority thread moved to suspended state and lower priority thread is continued its operation.

**Non-working scenario:**
When higher priority thread try to get already acquired mutex, higher priority thread is not moved to suspended state and keeps on running because of below "if" condition fail in _tx_thread_system_suspend() function.

/* Determine if preemption should take place. This is only possible if the current thread pointer is
not the same as the execute thread pointer AND the system state and preempt disable flags are clear. */
**if (_tx_thread_system_state[core_index] == ((ULONG) 0))**
**{**

The _tx_thread_system_state variable value is 1. So this condition is failed and higher priority thread is not moved to suspended state and keeps on running. Because the _tx_mutex_get() function return TX_SUCCESS after this condition failure.

We need proper working of mutex get function and higher priority thread should move to suspended state, if mutex is already acquired by other thread.

It is a showstopper issue for our project. Any solution to this problem is really helpful.

Thanks,
Devaraj Sivaprakasam.

fdesbiens, 2024-11-13:
Hi @devaraj-siva. 

Thank you for the detailed bug report. I will check with the other committers to see if anyone can help you.

Out of curiosity, do you see the same behaviour when using the Arm GNU compiler? What about different hardware platforms?

billlamiework, 2024-11-14:
Hello Devaraj,
This is an interesting problem. Theoretically, the only way _tx_thread_system_state[core_index] can be non-zero is if the API call is being made from an ISR on the core identified by "core_index". Is there any chance the tx_mutex_get call is being made from an ISR instead of a higher-priority thread?  Another possibility is that the _tx_thread_system_state structure is somehow being corrupted.
Best regards,
Bill  (blamie@px5rtos.com)

devaraj-siva, 2024-11-21:
Hi @fdesbiens ,
We verified the scenario. Both mutex get function is called from thread environment. We confirmed this by verifying the **thread_ptr** properties such as tx_thread_name inside **_tx_mutex_get** function. It reflects the respective thread name.
We didn't call any mutex get function from interrupt. 

If _tx_thread_system_state structure is corrupted, this behaviour will occur every time. But this behaviour is not occurred, if we reduced the frequent usage of mutex get function. For example, we used mutex get and set functions inside **printf** function. When we put lot of printf from different threads, this problem is occurred. But when we removed the printf statements(debug prints), this problem is not occurred.

So I suspect, the problem in mutex handling.

billlamiework, 2024-11-21:
Hi @devaraj-siva,

 In addition to looking at the _tx_thread_current_ptr[current_core] entry to see if the call is being made from an ISR, you should also simultaneously look at the _tx_thread_system_state[current_core] structure to see if an ISR is present. In other words, the current thread structure alone doesn't mean an ISR is not present.

The mutex code itself is quite extensively tested, so I don't expect a mutex problem. I'm thinking there is something wrong with this particular port of ThreadX. In addition to corruption possibilty, if there is a bug in this port that examines _tx_thread_system_state[current_core] of an incorrect core (perhaps after the thread is moved to a different core in the middle of the processing) this could cause a behavior like this.

A couple questions/ideas:

1. Do you have hardware trace capabilities?
2. Do you see that the thread making the printf call being switched to a different core when the problem occurs?
3. Does the mutex have priority inheritance enabled?
4. It would be best to have a smaller code test maybe with "n" threads all just competing on a single mutex to see if you can generate the error faster and with less code involved.

I have some additional ideas, please reach out to me at blamie@px5rtos.com and we can discuss further.

fdesbiens, 2024-12-05:
@devaraj-siva, Can you please provide feedback to @billlamiework? Without additional information, I will close this issue soon.

sushovan-i21377, 2024-12-05:
Hi @fdesbiens 
  We are working on this issue and have some feedback regarding TthreadX implementation of tx_mutex_get(). We will get back to you by EOD tomorrow,.

sushovan-i21377, 2024-12-06:
Hi @fdesbiens  , @billlamiework 
   Appreciate your feedback for this issue. We have seen one of the ISR is calling ThrreadX API and that creates the issue. We also observed the _tx_thread_system_state[0] value become 1 and that causes the the thread not going to suspend, We have fixed at out end.
   _However , this issue also brings out some interesting observation, I hope that need your immediate observation. The return value from _tx_mutex_get() return 0 irrespective of (1) the mutex is obtained , (1) thread gets suspend while waiting for mutex , and (3) the thread is not able to get mutex due to _tx_thread_system_state[0] =1. All three scenarios , return value is numerical zero._ 
  This is bacause  of the line 
                                             status =  thread_ptr -> tx_thread_suspend_status 

Only two places the return value from _tx_mutex_get()  is not 0 which are not part of the three cases above.

Had the return values been different in those three cases, the issue would have been avoided in the code , because we are handling the thread mutex based on the return value,

Do you agree with our observation?

billlamiework, 2024-12-06:
Hi @devaraj-siva, @sushovan-i21377, 

Very glad to hear the problem has been solved and thanks for following up! 

One question:  was error checking disabled or the return value not checked in the ISR code that was erroneously calling the ThreadX API?  The API error checking should provide all the safeguards needed to prevent this situation.

As for your observation, when the internal call is made to _tx_thread_system_suspend the system design requires it to suspend the calling thread. Under normal operation, when the "status" is picked up from "thread_ptr -> tx_thread_suspend_status" after the suspension is lifted, ThreadX has already set this field in the thread control block to one of the following:  TX_SUCCESS (mutex was released), TX_NOT_AVAILABLE (timeout on the mutex suspension), or TX_WAIT_ABORTED (a call to tx_thread_wait_abort API).  This happens in various places in the code base. This is the normal operation.

The situation in your case happens when the internal call to  _tx_thread_system_suspend fails and results in the thread not actually suspending. Instead, the thread keeps running and simply returns the last value of "thread_ptr -> tx_thread_suspend_state".  We could initialize the status everywhere to something other than TX_SUCCESS before suspending, but that doesn't solve the bigger problem, since in this situation a thread is now running that isn't on the internal thread ready list and its control block is still linked on the mutex suspension list. My point here is the system is in an unrecoverable state here - well beyond the returned status.

fdesbiens, 2024-12-12:
I am closing this issue since you found a fix @sushovan-i21377. Feel free to reopen or create a new one as needed.

closed 2024-12-12 by fdesbiens

sushovan-i21377, 2024-12-13:
hi @fdesbiens 
yes. thank you for your support.