eclipse-threadx/threadx #419: Comment on posix condition variable implementation looks bogus

wucke13, 2024-11-20:
- COndition -> Condition
- "into a semaphore  a cast" ? With the double space between "semaphore" and 'a'?

https://github.com/eclipse-threadx/threadx/blob/485a02faec6edccef14812ddce6844af1d7d2eef/utility/rtos_compatibility_layers/posix/px_cond_signal.c#L92

jserv, 2024-11-21:
The comment can be refined as following:
```diff
--- a/utility/rtos_compatibility_layers/posix/px_cond_signal.c
+++ b/utility/rtos_compatibility_layers/posix/px_cond_signal.c
@@ -89,7 +89,7 @@ TX_SEMAPHORE        *semaphore_ptr;
 UINT                 status;
 
     /* Get the condition variable's internal semaphore.  */
-    /* Simply convert the COndition variable control block into a semaphore  a cast */ 
+    /* Simply convert the condition variable control block into a semaphore using casting */
     semaphore_ptr = (&( cond->cond_semaphore ));
     if ( semaphore_ptr->tx_semaphore_suspended_count)
     {
```

wucke13, 2024-11-22:
Seems sensible to me!

fdesbiens, 2024-12-05:
@wucke13 Just to confirm I will make the change. I will take care of this next week, as I am on a trip right now.

fdesbiens, 2026-09-03:
Thanks for your patience, @wucke13, and thanks to @jserv for the suggested wording.

I opened https://github.com/eclipse-threadx/threadx/pull/694 to fix this. While looking at it, I found the same garbled sentence in three more files (`px_cond_init.c`, `px_cond_broadcast.c`, `px_cond_destroy.c`), with three different capitalisations of "condition", so the PR harmonises all four.

I will close this issue once the PR is merged.

fdesbiens, 2026-09-03:
PR #694 has been merged into `dev`. Thanks again @wucke13 for the report and @jserv for the suggested wording.

closed 2026-09-03 by fdesbiens